volumio/Volumio2 #10: Make volumio installable on an already running host

ning-yu, 2015-05-24:
This has been requested previously, reopening to track here.

jotak, 2015-05-24:
I think that even if it's not an "officially supported" feature, it would be nice to make it possible.

jomo, 2015-05-24:
Indeed this would be nice, especially for development.
Maybe we could have a list of requirements, so that could at least be manually set up.

jotak, 2015-05-24:
fyi some people already did it https://volumio.org/forum/volumio-ubuntu-t1299.html#p4917

volumio, 2015-05-24:
The installation instructions are a possible way. Providing a clean .deb install is not an option unfortunately...
As for development, using a supported platform is the best way, since is the only way to achieve the system compliance needed

arthurlutz, 2015-05-25:
Hi, am interested in this too. @volumio why is providing a clean .deb not an option ?

volumio, 2015-10-11:
It will take really a lot of time...

closed 2015-10-11 by volumio