volumio/Volumio2 #12: Failing to install through npm install on debian jessie

arthurlutz, 2015-05-25:
```
# npm install
/
> groove@2.3.4 install /root/local/volumio2/node_modules/groove
> node-gyp rebuild

make: Entering directory '/root/local/volumio2/node_modules/groove/build'
  CXX(target) Release/obj.target/groove/src/groove.o
In file included from ../src/groove.cc:5:0:
../src/player.h:6:33: fatal error: grooveplayer/player.h: No such file or directory
 #include <grooveplayer/player.h>
                                 ^
compilation terminated.
groove.target.mk:91: recipe for target 'Release/obj.target/groove/src/groove.o' failed
make: *** [Release/obj.target/groove/src/groove.o] Error 1
make: Leaving directory '/root/local/volumio2/node_modules/groove/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 3.16.0-4-686-pae
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /root/local/volumio2/node_modules/groove
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! groove@2.3.4 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the groove@2.3.4 install script.
npm ERR! This is most likely a problem with the groove package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls groove
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 3.16.0-4-686-pae
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! cwd /root/local/volumio2
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /root/local/volumio2/npm-debug.log
npm ERR! not ok code 0
```
I have a bunch of groove packages installed (in particular the -dev) : 

```
# dpkg -l *groove*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                      Version           Architecture      Description
+++-=========================-=================-=================-========================================================
ii  groovebasin               1.4.0-1           all               music player server with a web-based user interface
ii  libgroove-dev:i386        4.2.1-1           i386              audio dispatching library (development files)
ii  libgroove4:i386           4.2.1-1           i386              music player backend
ii  libgroovefingerprinter4:i 4.2.1-1           i386              acoustid fingerprinter for libgroove
ii  libgrooveloudness4:i386   4.2.1-1           i386              loudness scanner for libgroove
ii  libgrooveplayer4:i386     4.2.1-1           i386              hardware audio playback for libgroove
ii  node-groove               2.2.6-1           i386              bindings to libgroove - generic music player library - N
`

volumio, 2015-05-25:
Try to install build-essentials first

arthurlutz, 2015-05-25:
@volumio they are already installed.

It was failing to install *because* the system package was installed. So far removing the groove line in package.json works.  But then it fails on trying to start the service. 

Have a filed a bug against node-groove to try to find a solution : https://github.com/andrewrk/node-groove/issues/16

Aren't you missing in your install instructions some commands to install libgroove and it's developement headers ?

volumio, 2015-05-25:
Could be... Maybe removing node package and compiling it from source again will solve the issue

arthurlutz, 2015-05-25:
Nope, tried that too.

ning-yu, 2015-05-25:
Sorry this is my bad. I had added the Node Groove module as a requirement in order to do some testing on it. It requires libgroove to be already installed on the system, so that's probably why your install is breaking. None of the Volumio code currently uses groove, so I'm taking it out of the requirements.

closed 2015-05-25 by ning-yu