volumio/Volumio2 #14: Test Libgroove

ning-yu, 2015-05-28:
Evaluate the use of libgroove as a possible replacement for MPD. Reference https://github.com/andrewrk/libgroove/issues/88.

Maybe make a script which conducts double-blind listening trials of some high resolution music files?

jhoutman, 2015-05-28:
Would love to split test this ;-) not double blind perhaps.. :'(

volumio, 2015-05-28:
Yes, we'll do that but in a later stage of development (we should implement playback modules for both..)

andrewrk, 2015-07-07:
I've been working on [libsoundio](https://github.com/andrewrk/libsoundio). When this is done, libgroove will switch from SDL2 to using this. Once that is complete, I'll have written the code all the way down to the sound drivers, so I could provide a guaranteed bit perfect playback mode.

volumio, 2015-07-07:
That looks amazing! It could be used by Volumio as well... 
Taking a look at the APIs...

andrewrk, 2015-07-07:
> Taking a look at the APIs...

Please don't yet. I'm going to implement all the driver backends first before settling on an API. And I'm still deciding between [timer-based audio scheduling](http://0pointer.de/blog/projects/pulse-glitch-free.html) versus callbacks.

andrewrk, 2015-07-08:
One thing that I learned from researching audio backends is that you typically have a software (or hardware) mixer in order to allow multiple applications to make sound at once.

Consider if you have a song with 32-bit LE, Stereo, 48,000Hz and you want to play it back. So you open the hardware device with those parameters and begin playback. Now you open YouTube and play a video that has 16-bit LE 44,100 Hz audio. Typically your operating system will perform resampling and sample format conversion on YouTube's audio stream to make it work with the original parameters you opened the hardware device with. But what if YouTube opened the audio device first? In this case your song would not get bit perfect playback.

The only way to resolve this situation is to open the audio device in "greedy" mode, so you hog the sound card and control its parameters. This would make your music player prevent other applications from making sound.

volumio, 2015-07-08:
That is interesting, we've had this problem before, and in our case we just decided to allow one service to use the audio device at the time (when another audio process is started, it stops the previous one), but this makes sense in Volumio and is not viable in other situations...

andrewrk, 2015-07-17:
Here's what the libsoundio `list_devices` example program currently outputs for my laptop using the ALSA backend:

```
[nix-shell:~/dev/libsoundio/build]$ ./list_devices 
--------Input Devices--------

HDA Intel PCH, ALC892 Analog: Default Audio Device (default)
  name: default:CARD=PCH
  channel layouts:
    Stereo
  current layout: Stereo
  min sample rate: 4000
  max sample rate: 40265973
  current sample rate: 48000
  formats: unsigned 8-bit, signed 8-bit, signed 16-bit LE, signed 16-bit BE, unsigned 16-bit LE, unsigned 16-bit LE, signed 24-bit LE, signed 24-bit BE, unsigned 24-bit LE, unsigned 24-bit BE, signed 32-bit LE, signed 32-bit BE, unsigned 32-bit LE, unsigned 32-bit BE, float 32-bit LE, float 32-bit BE, float 64-bit LE, float 64-bit BE
  min buffer duration: 0.04266667 sec
  max buffer duration: 0.17066667 sec
  min period duration: 0.02133333 sec
  max period duration: 0.02133333 sec
  current period duration: 0.02133333 sec

HDA Intel PCH ALC892 Analog (raw)
  name: hw:0,0
  channel layouts:
    Stereo
  min sample rate: 44100
  max sample rate: 192000
  formats: signed 32-bit LE
  min buffer duration: 0.00016667 sec
  max buffer duration: 0.00266667 sec
  min period duration: 0.00008333 sec
  max period duration: 0.04266667 sec

HDA Intel PCH ALC892 Alt Analog (raw)
  name: hw:0,2
  channel layouts:
    Stereo
  min sample rate: 44100
  max sample rate: 192000
  formats: signed 32-bit LE
  min buffer duration: 0.00016667 sec
  max buffer duration: 0.00266667 sec
  min period duration: 0.00008333 sec
  max period duration: 0.04266667 sec

--------Output Devices--------

HDA Intel PCH, ALC892 Analog: Default Audio Device (default)
  name: default:CARD=PCH
  channel layouts:
    Stereo
  current layout: Stereo
  min sample rate: 4000
  max sample rate: 40265973
  current sample rate: 48000
  formats: unsigned 8-bit, signed 8-bit, signed 16-bit LE, signed 16-bit BE, unsigned 16-bit LE, unsigned 16-bit LE, signed 24-bit LE, signed 24-bit BE, unsigned 24-bit LE, unsigned 24-bit BE, signed 32-bit LE, signed 32-bit BE, unsigned 32-bit LE, unsigned 32-bit BE, float 32-bit LE, float 32-bit BE, float 64-bit LE, float 64-bit BE
  min buffer duration: 0.04266667 sec
  max buffer duration: 0.17066667 sec
  min period duration: 0.02133333 sec
  max period duration: 0.02133333 sec
  current period duration: 0.02133333 sec

HDA Intel PCH, ALC892 Digital: IEC958 (S/PDIF) Digital Audio Output
  name: iec958:CARD=PCH,DEV=0
  channel layouts:
    Stereo
  min sample rate: 32000
  max sample rate: 192000
  formats: signed 32-bit LE
  min buffer duration: 0.00016667 sec
[... 149 lines cut ...]
    Front Left, Front Right, Front Center, Front Left Center, Front Right Center
    Front Left, Front Right, Back Center, Front Left Center, Front Right Center
    Front Left, Front Right, LFE, Front Center, Back Left, Back Right
    Front Left, Front Right, LFE, Back Left, Back Right, Back Center
    Hexagonal
    Front Left, Front Right, Back Left, Back Right, Back Left Center, Back Right Center
    Front Left, Front Right, LFE, Front Center, Front Left Center, Front Right Center
    Front Left, Front Right, LFE, Back Center, Front Left Center, Front Right Center
    Front Left, Front Right, Front Center, Back Center, Front Left Center, Front Right Center
    Front Left, Front Right, Back Left, Back Right, Front Left Center, Front Right Center
  min sample rate: 32000
  max sample rate: 192000
  formats: signed 32-bit LE
  min buffer duration: 0.00004167 sec
  max buffer duration: 0.00066667 sec
  min period duration: 0.00002083 sec
  max period duration: 0.01066667 sec

11 devices found

```

So the first (default) output device is using ALSA dmix plugin to support any sample rate and any sample format. If you play back audio to this device you will not get bit perfect playback and you will share the device so that other applications can use the device simultaneously. If you choose a sample rate other than the "current" one then your audio will get resampled.

Note also the (raw) output device which is hw:0,0. This is the device that the default output device ultimately outputs to, but if you open this one in raw mode, you are the only application which can use the device until you close it. Note that it supports only one channel layout - stereo, and only one sample format - signed 32-bit LE. If you play any song which does not fit these requirements, you can not get bit perfect playback since the hardware does not support it. Fortunately sample format conversion is not super lossy like resampling is, and apparently sound cards support a wide range of sample rates.

volumio, 2015-07-17:
Hi Andrew, from what I see it's a truly compelling implementation! Love the idea of devices and raw devices. This allows to have great flexibility: conventient if you want just to play on a card, if you want bit perfect you can still have it under certain conditions.
I'm thinking that libsoundio could be a great fit for Volumio2. Do you have any ETA on the project? 
Also, what do you consider the best option for on-the-go output switch: a loopback audio device, FIFO or JACK ?

andrewrk, 2015-07-18:
> Do you have any ETA on the project?

I'm working on it full time. The roadmap is here: https://github.com/andrewrk/libsoundio/#roadmap
I keep that updated, so that can give you an indication of progress. What are your target platforms? Full Linux support will probably arrive before OSX and Windows.

> what do you consider the best option for on-the-go output switch: a loopback audio device, FIFO or JACK ?

When you say on the go output switch, do you mean allowing the user to switch the output from one device to another? What does FIFO mean in this context? PulseAudio supports this out of the box, btw.

andrewrk, 2015-09-06:
libsoundio 1.0.0 is released.  http://libsound.io/

Now I'm working toward libgroove 5.0.0 which is built on top of libsoundio instead of SDL2.

volumio, 2015-09-06:
Excellent!!!! Will give it a try now!

andrewrk, 2015-09-09:
Great, let me know how it goes for you.

Meanwhile, I'm working on the next major version of libgroove, 5.0.0. If you were still considering writing volumio on top of libgroove, consider following this issue: https://github.com/andrewrk/libgroove/issues/114

volumio, 2015-09-09:
Interesting! I think we ll complete a full mpd integration and then start with groove, a good way could also be let user decide which Be to use..

closed 2016-07-10 by volumio