volumio/Volumio2 #18: Setup of the dev environnement: error

tdemalliard, 2015-07-15:
Hello,
I am trying to setup the dev environnement with vmware and got an issue on ``npm install``. Here are the steps i follow:
* download vmware image at http://repo.volumio.org/Volumio2/DevTools/VolumioVM-VMWare.zip
* boot the vm with bridged network
* login from ubuntu 14.10 dekstop with ssh : ``ssh volumio@192.168.1.10``
* clone the git ``git clone https://github.com/volumio/Volumio2.git volumio/``
* npm install: ``cd volumio/ && npm install``

And got the following error:
````
volumio@volumio:~/volumio$ npm install
npm ERR! Linux 3.16.0-4-686-pae
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! node v0.12.0
npm ERR! npm  v2.5.1
npm ERR! code ETARGET

npm ERR! notarget No compatible version found: ip@'>=2.11.2 <3.0.0'
npm ERR! notarget Valid install targets:
npm ERR! notarget ["0.0.1","0.0.2","0.0.3","0.0.4","0.0.5","0.1.0","0.2.0","0.3.0","0.3.1","0.3.2","0.3.3"]
npm ERR! notarget 
npm ERR! notarget This is most likely not a problem with npm itself.
npm ERR! notarget In most cases you or one of your dependencies are requesting
npm ERR! notarget a package version that doesn't exist.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/volumio/volumio/npm-debug.log
````

npm-debug.log: http://pastebin.com/LZCefXWQ

Thank you.

volumio, 2015-07-15:
My bad, should be fixed now

tdemalliard, 2015-07-15:
it is fixed.
Merci.

closed 2015-07-15 by tdemalliard