volumio/Volumio2 #19: Documentation with jsDoc

hans-sperling, 2015-07-20:
Why don't you comment your code with thy syntax of [jsDoc](http://usejsdoc.org) so you build yout dev-docs every time you develop or change something.

volumio, 2015-07-20:
Yep I just thought today about an inline doc gererator. Also we could use docco http://jashkenas.github.io/docco/

Or I'm open to any suggestion...

hans-sperling, 2015-07-20:
In my oppinion you should use a *well known* documentation tool like *jsDoc*.
I don't know *Docco* but I read **"Docco is a quick-and-dirty documentation generator [...]"** in the first line of their project.

volumio, 2015-07-20:
Ok, I'll trust you on this. Since this is a collaborative project, want to hear the opinion of other contributors as well. 
Giving a look to jsdoc now, a way to generate hooks to github like redthedocs would be a plus of course (don't know if it has something like this)

plesatejvlk, 2015-07-20:
Hi, I've used JSDoc with several enterprise projects and I would definitely recommend it. Since it is a de-facto standard nowadays for javascript source documentation, it's support (click-trough, dot hints, even type checks) in various IDEs is already there.

For the generation part - I've personally found the JSDoc3 project (https://github.com/jsdoc3/jsdoc) quite handy. It is also very easy to modify the templates.

hans-sperling, 2015-07-20:
Oh, I meant jsDoc3, too.

volumio, 2015-07-20:
That seems really excellent.
First candidate for the API docs...

volumio, 2015-07-20:
So, jsdoc feels like the best option.
What about a feature that allows automatic updates of the docs? 

Or, initially something like:
https://github.com/thlorenz/jsdoc-githubify

To document just APIs on this wiki

hans-sperling, 2015-07-22:
You can create/update the docs:
-* manually via console,
-* by triggering your IDE
-* on a new build or release by tool like [grunt](http://gruntjs.com/) or [gulp](http://gulpjs.com/)

philkunz, 2015-10-29:
Before putting any more work in here: Consider moving to TypeDoc with TypeScript.

closed 2017-06-02 by volumio