volumio/Volumio2 #34: npm install fails

balbuze, 2015-08-18:
When doing npm install, it return errors 
npm install
npm WARN package.json Volumio2@2.0.0-alpha4 No license field.
-
> mdns@2.2.9 install /volumio/node_modules/mdns
> node-gyp rebuild

make: Entering directory '/volumio/node_modules/mdns/build'
  CXX(target) Release/obj.target/dns_sd_bindings/src/dns_sd.o
In file included from ../src/dns_sd.cpp:1:0:
../src/mdns.hpp:32:20: fatal error: dns_sd.h: No such file or directory
 #include <dns_sd.h>
                    ^
compilation terminated.
dns_sd_bindings.target.mk:136: recipe for target 'Release/obj.target/dns_sd_bindings/src/dns_sd.o' failed
make: *** [Release/obj.target/dns_sd_bindings/src/dns_sd.o] Error 1
make: Leaving directory '/volumio/node_modules/mdns/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Linux 3.18.6+
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /volumio/node_modules/mdns
gyp ERR! node -v v0.12.6
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok 
npm ERR! Linux 3.18.6+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install"
npm ERR! node v0.12.6
npm ERR! npm  v2.11.2
npm ERR! code ELIFECYCLE

npm ERR! mdns@2.2.9 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the mdns@2.2.9 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the mdns package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls mdns
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /volumio/npm-debug.log

If I do "node index.js" it retruns
/volumio# node index.js
module.js:338
    throw err;
          ^
Error: Cannot find module './http/index.js'
    at Function.Module._resolveFilename (module.js:336:15)
    at Function.Module._load (module.js:278:25)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (/volumio/index.js:1:85)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Function.Module.runMain (module.js:501:10)

No webUI and if I look in the process, node is eating plenty of cpu (on rpi B+ a total of 88% of cpu is used).
What am I doing wrong ?

jomo, 2015-08-18:
Have you seen [node_mdns Installation](https://github.com/agnat/node_mdns#installation)?

volumio, 2015-08-18:
Before insalling:
apt-get install libavahi-compat-libdnssd-dev

balbuze, 2015-08-18:
I erase the /volumio folder and restart from the beginning.
With "apt-get install libavahi-compat-libdnssd-dev" installation works much better just a small message a the beginning 
 "npm install
npm WARN package.json Volumio2@2.0.0-alpha4 No license field.
npm WARN optional dep failed, continuing fsevents@0.3.8
npm ERR! registry error parsing json
"
But I still can't reach the UI. Only ssh works. Node is working and eating 98% of the cpu and its PID changes every 10 sec (about).

volumio, 2015-08-18:
There's clearly something wrong...
Which image are you using?

jomo, 2015-08-18:
Can you paste the full npm log? I suspect there's something that tells us why it's failing to parse the JSON.

balbuze, 2015-08-18:
Image is VolumioAlpha4-2015-08-02PI
sorry but where am I suppose to find npm log ?

jomo, 2015-08-18:
Are those 3 lines from your previous comment all it spits out? :confused:

balbuze, 2015-08-18:
The full output is
"npm install
npm WARN package.json Volumio2@2.0.0-alpha4 No license field.
npm WARN optional dep failed, continuing fsevents@0.3.8
npm ERR! registry error parsing json
 
> mdns@2.2.9 install /volumio/node_modules/mdns
> node-gyp rebuild

make: Entering directory '/volumio/node_modules/mdns/build'
  CXX(target) Release/obj.target/dns_sd_bindings/src/dns_sd.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/dns_service_browse.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/dns_service_enumerate_domains.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/dns_service_get_addr_info.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/dns_service_process_result.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/dns_service_ref.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/dns_service_ref_deallocate.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/dns_service_ref_sock_fd.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/dns_service_register.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/dns_service_resolve.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/mdns_utils.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/network_interface.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/socket_watcher.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/txt_record_ref.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/txt_record_create.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/txt_record_deallocate.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/txt_record_set_value.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/txt_record_get_length.o
  CXX(target) Release/obj.target/dns_sd_bindings/src/txt_record_buffer_to_object.o
  SOLINK_MODULE(target) Release/obj.target/dns_sd_bindings.node
  COPY Release/dns_sd_bindings.node
make: Leaving directory '/volumio/node_modules/mdns/build'

> utf-8-validate@1.1.0 install /volumio/node_modules/socket.io/node_modules/engine.io/node_modules/ws/node_modules/utf-8-validate
> node-gyp rebuild

make: Entering directory '/volumio/node_modules/socket.io/node_modules/engine.io/node_modules/ws/node_modules/utf-8-validate/build'
  CXX(target) Release/obj.target/validation/src/validation.o
  SOLINK_MODULE(target) Release/obj.target/validation.node
  COPY Release/validation.node
make: Leaving directory '/volumio/node_modules/socket.io/node_modules/engine.io/node_modules/ws/node_modules/utf-8-validate/build'

> bufferutil@1.1.0 install /volumio/node_modules/socket.io/node_modules/engine.io/node_modules/ws/node_modules/bufferutil
> node-gyp rebuild

make: Entering directory '/volumio/node_modules/socket.io/node_modules/engine.io/node_modules/ws/node_modules/bufferutil/build'
  CXX(target) Release/obj.target/bufferutil/src/bufferutil.o
  SOLINK_MODULE(target) Release/obj.target/bufferutil.node
  COPY Release/bufferutil.node
make: Leaving directory '/volumio/node_modules/socket.io/node_modules/engine.io/node_modules/ws/node_modules/bufferutil/build'

> bufferutil@1.1.0 install /volumio/node_modules/socket.io/node_modules/socket.io-client/node_modules/engine.io-client/node_modules/ws/node_modules/bufferutil
> node-gyp rebuild

make: Entering directory '/volumio/node_modules/socket.io/node_modules/socket.io-client/node_modules/engine.io-client/node_modules/ws/node_modules/bufferutil/build'
  CXX(target) Release/obj.target/bufferutil/src/bufferutil.o
  SOLINK_MODULE(target) Release/obj.target/bufferutil.node
  COPY Release/bufferutil.node
make: Leaving directory '/volumio/node_modules/socket.io/node_modules/socket.io-client/node_modules/engine.io-client/node_modules/ws/node_modules/bufferutil/build'

> utf-8-validate@1.1.0 install /volumio/node_modules/socket.io/node_modules/socket.io-client/node_modules/engine.io-client/node_modules/ws/node_modules/utf-8-validate
> node-gyp rebuild

make: Entering directory '/volumio/node_modules/socket.io/node_modules/socket.io-client/node_modules/engine.io-client/node_modules/ws/node_modules/utf-8-validate/build'
  CXX(target) Release/obj.target/validation/src/validation.o
  SOLINK_MODULE(target) Release/obj.target/validation.node
  COPY Release/validation.node
make: Leaving directory '/volumio/node_modules/socket.io/node_modules/socket.io-client/node_modules/engine.io-client/node_modules/ws/node_modules/utf-8-validate/build'

> leveldown@0.10.5 install /volumio/node_modules/level/node_modules/leveldown
[... 155 lines cut ...]
└── rimraf@2.4.2 (glob@5.0.14)

fast.js@0.1.1 node_modules/fast.js

musicbrainz@0.2.5 node_modules/musicbrainz
├── timetrickle@0.1.1
├── request@1.9.9
└── xml2js@0.1.14 (sax@1.1.1)

mdns@2.2.9 node_modules/mdns
├── bindings@1.2.1
└── nan@1.8.4

socket.io@1.3.6 node_modules/socket.io
├── debug@2.1.0 (ms@0.6.2)
├── has-binary-data@0.1.3 (isarray@0.0.1)
├── socket.io-adapter@0.3.1 (object-keys@1.0.1, debug@1.0.2, socket.io-parser@2.2.2)
├── socket.io-parser@2.2.4 (isarray@0.0.1, debug@0.7.4, component-emitter@1.1.2, benchmark@1.0.0, json3@3.2.6)
├── engine.io@1.5.2 (base64id@0.1.0, debug@1.0.3, engine.io-parser@1.2.1, ws@0.7.2)
└── socket.io-client@1.3.6 (to-array@0.1.3, indexof@0.0.1, component-bind@1.0.0, debug@0.7.4, backo2@1.0.2, object-component@0.0.3, component-emitter@1.1.2, has-binary@0.1.6, parseuri@0.0.2, engine.io-client@1.5.2)

level@0.18.0 node_modules/level
├── level-packager@0.18.0 (levelup@0.18.6)
└── leveldown@0.10.5 (bindings@1.2.1, nan@1.8.4)
"

volumio, 2015-08-18:
I just replicated the procedure and everything is fine here. 
Try deleting /volumio and recloning it 
Try also as root...

jomo, 2015-08-18:
```
npm ERR! registry error parsing json
```

That json error is really weird. The rest of the npm log looks fine, though.

balbuze, 2015-08-18:
I have removed /volumio folder and restart the procedure.
After installing, if I do node index.js I get 
"node index.js
Loading plugins from folder /volumio/app/plugins/
Loading plugins from folder /plugins
[1439932166605] Loading plugin "system"...
No id defined. Creating one
[1439932167204] Loading plugin "network"...
[1439932168044] Loading plugin "services"...
[1439932168364] Loading plugin "volumiodiscovery"...
*** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
*** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
[1439932169240] Loading plugin "upnp"...
[1439932172094] Loading plugin "dirble"...
Plugin example_plugin is not enabled
[1439932172147] Loading plugin "mpd"...
[1439932176083] Loading plugin "spop"...
Plugin mpdemulation is not enabled
[1439932176705] Loading plugin "websocket"...
events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: listen EADDRINUSE
    at exports._errnoException (util.js:746:11)
    at Server._listen2 (net.js:1156:14)
    at listen (net.js:1182:10)
    at Server.listen (net.js:1267:5)
    at Function.app.listen (/volumio/node_modules/express/lib/application.js:556:24)
    at Object.<anonymous> (/volumio/index.js:6:29)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
"
I will reflash my sd card tomorow and retry...

jomo, 2015-08-18:
> `Error: listen EADDRINUSE`

That means it can't start the server because the port is in use. I guess you just have it running already? Check with `pgrep -l node`

volumio, 2015-08-18:
yes it means you have the server already running, it's normal

if you want to see the new version.
just do

killall node

balbuze, 2015-08-19:
So I have reflashed my SD card
Boot RPI B with it- everthing works as expected - webUI is reachable
Next step 
 apt-get install libavahi-compat-libdnssd-dev
and I have ereased /volumio
then 
"git clone https://github.com/volumio/Volumio2.git volumio
cd volumio
npm install"
Still have some error at the beginning 
" npm install
npm WARN package.json Volumio2@2.0.0-alpha4 No license field.
npm WARN cannot run in wd Volumio2@2.0.0-alpha4 git submodule init && git submodule update (wd=/volumio)
npm WARN optional dep failed, continuing fsevents@0.3.8
...."
When finished 
"node killall"
returns the following 
" node killall
module.js:338
    throw err;
          ^
Error: Cannot find module '/volumio/killall'
    at Function.Module._resolveFilename (module.js:336:15)
    at Function.Module._load (module.js:278:25)
    at Function.Module.runMain (module.js:501:10)
    at startup (node.js:129:16)
    at node.js:814:3
"
The webUI displays "Cannot GET/"
The dev console is reachable
There is something missing in the procedure or what ?

volumio, 2015-08-19:
killall node

closed 2015-08-25 by volumio

Dermiste, 2015-09-12:
I had the same problem and got it solved by doing what was explained here. The webUI seems to provide angular error and I get this when I run node index.js:
```javascript
Loading plugins from folder /home/pi/volumio/app/plugins/
Loading plugins from folder /plugins
[1442064929388] Loading plugin "system"...
[1442064929543] Loading plugin "network"...
[1442064929889] Loading plugin "networkfs"...
[1442064930135] Loading plugin "services"...
[1442064930301] Loading plugin "volumiodiscovery"...
*** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
*** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
Started advertising...
[1442064930806] Loading plugin "upnp"...
[1442064931731] Starting Upmpd Daemon
[1442064931839] Loading plugin "dirble"...
Plugin example_plugin is not enabled
[1442064931890] Loading plugin "mpd"...
[1442064933392] ControllerMpd::StartedWatchService
[1442064933398] Loading plugin "spop"...
[1442064933594] ControllerSpop::loadTracklistFromDB
Loading Spop tracklist from DB...
Plugin mpdemulation is not enabled
[1442064933702] Loading plugin "websocket"...
[1442064937738] CoreMusicLibrary::loadLibraryFromDB
Loading library from DB...
[1442064937842] CorePlaylistFS::loadPlaylistsFromDB
Loading playlists from DB...
[1442064937855] BOOT COMPLETED
Express server listening on port 3000
[1442064937909] CoreStateMachine::resetVolumioState
[1442064937914] CoreStateMachine::getcurrentVolume
[1442064937917] CoreCommandRouter::volumioRetrievevolume
[1442064937980] CoreStateMachine::updateTrackBlock
[1442064937996] CorePlayQueue::getTrackBlock
[1442064938507] ControllerSpop::pushError(Error: Error reading DB: NotFoundError: Key not found in database [tracklist])
[1442064938520] CoreMusicLibrary::pushError(Error: Error reading DB: NotFoundError: Key not found in database [library])
MPD error: 
{ [Error: connect ECONNREFUSED]
  code: 'ECONNREFUSED',
  errno: 'ECONNREFUSED',
  syscall: 'connect' }
SPOP command error:
{ [Error: connect ECONNREFUSED]
  code: 'ECONNREFUSED',
  errno: 'ECONNREFUSED',
  syscall: 'connect' }
SPOP status error:
{ [Error: connect ECONNREFUSED]
  code: 'ECONNREFUSED',
  errno: 'ECONNREFUSED',
  syscall: 'connect' }
The following error occurred while starting SPOPD: Error: Command failed: /bin/sh: 1: spopd: not found

[1442064938657] Upmpcli error: Error: Command failed: /bin/sh: 1: upmpdcli: not found

Changing my name to raspberrypi
[1442064938871] mDNS: Found device volumio
[1442064938920] VolumeController:: Volume=80 Mute =false
[1442064938927] CoreStateMachine::updateVolume80
[1442064938932] CoreStateMachine::getState
[1442064938940] CoreStateMachine::pushState
[1442064938945] CoreCommandRouter::volumioPushState
[1442064938951] InterfaceWebUI::pushState
[1442064944717] mDNS: Found device raspberrypi
```

volumio, 2015-09-12:
At this point of development it would be really hard for you to set a complete working environment. We are almost at the point of having a first beta image... If you can wait a little...

Dermiste, 2015-09-12:
The AirPlay feature does not seem to work so far also ... but of course I can wait :) Keep up the good work !

volumio, 2015-09-12:
Yep this doesn't work yet, it will be one of the last things we'll tackle...