volumio/Volumio2 #38: subsonic music server

derEremit, 2015-08-21:
I know you're just doing ground work right now, but as a long time subsonic user I've just seen they integrated with Sonos devices, and since the have a well documented api, it shoud be "fairly" easy to implement.
http://www.subsonic.org/pages/sonos.jsp
And as subsonic supports lossless streaming I think it's a perfect fit.
http://www.subsonic.org/pages/api.jsp

If there's noone willing to do it you could at least verify if it is possible or point me in directions where to start. Got a little node experience myself!

closed 2016-07-10 by volumio

Bjoern3003, 2016-07-22:
Any News here? I am also interested in subsonic Support für V2.

Perhaps, this repo could help for Volumio Plugin: https://github.com/rattboi/mopidy-subsonic/tree/master/mopidy_subsonic

MarkAdamson, 2018-06-08:
Baffling response from the volumio devs here, and in the "Suggest a feature" [thread](https://volumio.org/forum/support-for-subsonic-t5753.html).

FWIW, Subsonic is not a 'piracy tool', but simply a tool for listening to music stored remotely, something which Volumio _already does_ via other protocols. I use Subsonic personally to have access to all of my legally acquired and owned music wherever I am.

Subsonic support in Volumio would mean I don't have to scan my music collection twice and store two different databases - Subsonic could be a drop-in replacement for Volumios database.

Also, the suggestion of legal complications is patently ludicrous. Plex have built a successful business on a very similar technology, and take the stance that what people use the software for is their own responsibility.

@derEremit, I have no doubt this would be possible, though I imagine quite difficult unless you're very familiar with the Volumio code and/or Subsonic api. And if Volumio devs continue with this stonewall approach, you'll be forking and developing without their support.

Very disappointing guys.

volumio, 2018-06-12:
@MarkAdamson we focused a lot on creating the plugin facility in order to allow anyone to add  the functionality he thinks are cool, while we as core team take care most of mantaining the core code and improving basic functionalities. So, no stonewall approach here, but rather optimizing our resources.

That being said, I think what subsonic offers is really cool, and we're working hard to bring the same to Volumio in the future

grravity, 2020-04-01:
Was curious if this was ever implemented, I don't have a Pi to test out Volumio but don't want to have to deal with Volumio and Airsonic (Fork of Subsonic) alongside each other. As Mark said:

>Subsonic support in Volumio would mean I don't have to scan my music collection twice and store two different databases - Subsonic could be a drop-in replacement for Volumios database.

It'd be nice to be able to just pull music from Subsonic/Airsonic.
Was just curious if this was ever implemented, it's been 2 years since the last comment here. Again, I can't see for myself as I don't have a Pi to test it on, and figured it'd just be easier to ask.

volumio, 2020-04-02:
There is a subsonic plugin available.

grravity, 2020-04-02:
@volumio thank you