volumio/Volumio2 #40: Solve DNS uncaught error event

volumio, 2015-08-21:
This is the error:

events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: dns service error: unknown
    at Error (native)
    at MDNSService.on_resolver_done (/volumio/node_modules/mdns/lib/resolver_sequence_tasks.js:19:30)
    at SocketWatcher.MDNSService.self.watcher.callback (/volumio/node_modules/mdns/lib/mdns_service.js:18:40)

It's quite random

jomo, 2015-08-22:
Might be helpful
- https://github.com/agnat/node_mdns/issues/97
- https://github.com/agnat/node_mdns/issues/104

volumio, 2015-08-22:
Thanks. We're gonna add an error handler to it.

Also, I noticed that we still have unhandled errors in some places in the system. Should start to place error handling while we code, as a good practice

fanciulli, 2015-08-25:
Error catching has been introduced. This will restart the advertising or the browse functionality

fanciulli, 2015-09-03:
Closing this issue since error never show up again after update

closed 2015-09-03 by fanciulli