volumio/Volumio2 #44: Missing Spotify account configuratiom

fanciulli, 2015-08-25:
Spotify configuration cannot be entered in the current configuration Ui. A section should contain services configurations.

glucaci, 2016-04-19:
I've just made something in this direction. Where can I create a PR, on dev or master?

volumio, 2016-04-19:
On Dev, but please wait since we're using a new method to do this...

glucaci, 2016-04-19:
Maybe I can help, if you can describe your ideea and if nobody is working on it already.

volumio, 2016-04-19:
I would love an helping hand on this... But we're in the design process still of the whole plugin management system, so there's nothing definitive yet...
I know it's not the best answer... Hopefully for middle of next week we'll have something working , and then you could jump in and give us your feedback and contribute.. I'll keep you  posted..
What do you think?

glucaci, 2016-04-20:
Ok, I understand the situation. Till then, I will just try to take a look on Issues that are not so infrastructure linked.

apinner, 2016-06-27:
Fixed with the spotify plugin

closed 2016-06-27 by apinner