volumio/Volumio2 #49: WebUI or dev console

balbuze, 2015-08-30:
Hi!
Sorry to ask, but even with a fresh install, I can't connect to webUI (still known problem?)
dev console was ok until I reboot...and now no more access.

volumio, 2015-08-30:
Try from scratch 

systemctl disable volumio.service
rm -rf /volumio
git clone https://github.com/volumio/Volumio2.git /volumio
cd /volumio
npm install --unsafe-perm

Now to start everything you need to lauch manually with
cd /volumio
node index.js

balbuze, 2015-08-30:
Thank you !
dev console is working...
What about webUI ? (blank page)

mpasteur, 2015-11-20:
by default the submodule (in http/www) is not retrieved when you clone the repo. You have to get it by 'git clone --recursive https://github.com/volumio/Volumio2.git volumio'

balbuze, 2015-11-20:
@mpasteur thank you, but it was an old problem...

mpasteur, 2015-11-20:
you're right but i've just encountered this problem... So it can be usefull for the Internet :-)

balbuze, 2015-11-20:
ok ;-)

closed 2016-04-03 by apinner