volumio/Volumio2 #59: Use Vagrant

jomo, 2015-09-12:
In case you're not familiar with [Vagrant](https://www.vagrantup.com/), it basically lets you set up a VM that you can configure to contain all dependencies. It syncs your (git) project folder with the VM so you can make changes on your machine and they affect the VM immediately.
You can forward the network ports required for the project and connect to the VM via SSH.

This creates a "lightweight, reproducible, and portable development environment".
I believe it could make development a lot easier.

volumio, 2015-09-12:
I absolutely agree that Volumio needs easier ways to develop. That is why so far we have:
- VMWare VM image
- VirtualBox VM image
- Docker image 

Adding Vagrant could be a good thing. If you want to tackle this ...

jomo, 2015-09-12:
I tried to figure out how to do this, and we would have to [package the VMs](https://docs.vagrantup.com/v2/virtualbox/boxes.html) into a `.box` file, which is something that should be done with [volumio/build](https://github.com/volumio/Build).

I'll try to do that soon™

volumio, 2015-09-12:
Lol love the last sentece... sounds familiar...
Simply add the needed lines at the end of https://github.com/volumio/Build/blob/master/scripts/x86.sh

Probably the best thing would be add also command line parameters (which are inherited trought the whole scripting chain) to build.sh

Let me know if you need help with this!

EMCP, 2015-10-27:
You could embed a `Vagrantfile` in this project, which described how to pull in a fresh debian/ubuntu OS.. install the dependencies, create the user, and run `nodejs`

problem I'd have is that right now when I run `npm install` it doesn't work in raspberry pis running the latest debian/raspbian.

making a .box doesn't really solve that, and may infact hide developers from seeing the current state of the OS/environment for other devs

volumio, 2015-11-18:
I agree on your points, the fact is that we're so busy developing the system...
We have this build repo:
https://github.com/volumio/Build

Want to help with vagrant box creation?

EMCP, 2015-11-20:
It appears that repo is using arch linux correct?  I have never made a box for Arch but I could give it a shot if that's the easiest for the team

volumio, 2015-11-21:
Nope is debian

closed 2016-04-03 by apinner