volumio/Volumio2 #64: npm install error on cookie-parser

EMCP, 2015-09-30:
I ran `npm install` as non-sudo user.  Seems `cookie-parser` failed to install

I know you don't run the NPM, I'm filing a bug there.. but just giving a heads up.

EMCP, 2015-09-30:
after trying to run it anyway.. looks like express didn't install.

volumio, 2015-09-30:
did you give proper permission to volumio folder? In which system are you installing?

EMCP, 2015-10-01:
this is a Raspberry Pi running Debian aka "Raspbian" ,

I will look again at the instructions again, I just created folders under my user's home dir

volumio, 2015-10-07:
UI works on Volumio distro, since it requires lot of dependencies

closed 2015-10-07 by volumio

EMCP, 2015-10-12:
I've re-read and followed the instructions.. same problem is occurring however.

volumio, 2015-10-12:
Are you using raspbian or Volumio2 distro?

EMCP, 2015-10-12:
`Raspbian` , I have a need to do that due to some customizations on the wifi routing

volumio, 2015-10-12:
So, can't help. The whole system is made on purpose.
Also, this is a raspbian problem (probably node's version) or lack of compilers

EMCP, 2015-10-18:
may I ask, what OS is this distro based on? Debian?

volumio, 2015-10-19:
Yes, debian. Built with multistrap