volumio/Volumio2 #66: Support - Bluetooth speaker

osos, 2015-10-05:
I am planning to set up a kids music player. As for speakers I want to stream directly to bluetooth enabled speakers.

Please implement support for this.

jomo, 2015-10-05:
That would be possible when #32 is implemented

osos, 2016-04-01:
Any news on this one?

Will it be possible with Volumio2

osos, 2016-04-01:
Maybe this could be achieved via ALSA ?

Inspiration:
http://mygeeks014.blogspot.dk/2015/01/audio-streaming-to-bluetooth-speaker.html

Would it work?

osos, 2016-04-30:
With Mopidy on Raspberry PI i managed to use my bluetooth speaker following the suggestions here: https://github.com/ev3dev/ev3dev/issues/198

Using Pulseaudio and bluez.

Kittyjuls, 2016-12-29:
I used the same process as above to get bluetooth working, until it broke last week. Doesn't seem to work anymore with the latest build, at least not for me. Bluetooth connects but no audio playback. 
Are there any plans for bluetooth support for Volumio in the pipeline? Appreciate development is done by a dedicated few that sacrifice their time to make all this possible.

UnleashSpirit, 2020-10-20:
Damn I thought it was available by default as all last Rpi have built-in bluetooth and I fall into this issue :'(

Any information about this huge enhancement feature :) ?

timothyjward, 2020-10-21:
> Any information about this huge enhancement feature :) ?

Bluetooth playback (e.g. from your phone to Volumio) has existed for a while. It’s available to all virtuoso and superstar users.
Is that what you want, or are you trying to play from Volumio to a Bluetooth device?

UnleashSpirit, 2020-10-21:
Hi,
Yeah I saw that from phone to volumio exists but only in the paid versions.
My wish is the other way, from volumio to bluetooth equipment :)

volumio, 2020-10-21:
Hi, luckily Tim has made a very cool development in Volumio which will allow this feature  to work in future releases.
So we will do it, but it will take time

marcomow, 2020-11-28:
Hi @volumio, do you have a release timeline? I'm evaluating a paid subscription, and having output to bluetooth is the main factor, thanks!