volumio/Volumio2 #76: Move from JS to TypeScript

philkunz, 2015-10-29:
As volumio is a bigger project I would propose moving to TypeScript with a gulp compile pipeline.

fanciulli, 2016-02-16:
Seems interesting but i am wondering what the impact would be for Volumio.

philkunz, 2016-02-16:
What should the impact be? It would allow better structuring of the code, would give access to things like easy classes without breaking support for ES5.
Errors that are not caught with TypeScript can be traced to their origin with source maps.

fanciulli, 2016-02-16:
Can we start using it leaving the old code in place? I think it is probably possible..

philkunz, 2016-02-16:
Since TypeScript is a superset of JavaScript that should work.

philkunz, 2016-02-16:
I've created a tool that helps with writing TypeScript for (npm) modules:
https://www.npmjs.com/package/npmts
I'll add automatic source map integration later this week.

If there are special requirements for volumio that you'd like to see integrated, let me know.

apinner, 2016-04-03:
Anything happening on this @philkunz 

Reopen if something is still happening.

closed 2016-04-03 by apinner

philkunz, 2016-04-04:
@apinner Well, I'm not putting any more work into this project, until the license is clarified. Then I'll be happy to fork, do the switch and hand in a PR. Until then forking is to be considered a grey area in law.