volumio/Volumio2 #77: Make volumio a requireable npm module.

philkunz, 2015-10-29:
I would suggest making volumio an requireable npm module with automated deploy using travis for extending volumio.

I have registered the "volumio" package name on npm before it is snatched up by some project unrelated package.

fanciulli, 2015-10-29:
HI,

i don't see why Volumio should be an npm package since it is not a library. Do you see any use case?

Massimiliano

philkunz, 2015-10-29:
Hi,

* it allows installation of volumio through npm, which is the default way to install things in nodejs.
* you could integrate volumio as part of other projects... say you have a gulp build pipeline with specific customizations and at the end you want to start volumio on a special port with special login data for Spotify etc : Simply create an options object and pass it to volumio upon require.

volumio, 2015-11-18:
Unfortunately this goes out of the scope of the project, as of now...

closed 2015-11-18 by volumio