volumio/Volumio2 #91: volumio.service start

gkkpch, 2015-11-16:
I can't get the volumio.service to start with systemd as **volumio.log** can't be created in **/var/log** due to permissions being denied. My first thought was to create **/var/log/volumio** in the build script, giving ownership to volumio:volumio (similar to what the mpd post install does). After that, both the logfile path strings in **/volumio/app/index.js** could be changed to /var/log/**volumio**/volumio.log).
This won't work when /var/log is going to be on tmpfs and disappears. Any suggestions other than doing this in rc.local?

volumio, 2015-11-16:
On the PI we use FSTAB to give proper permission to /var/log:
https://github.com/volumio/Build/blob/master/scripts/raspberryconfig.sh

You can achieve the same by putting this in fstab:
tmpfs   /var/log                tmpfs   size=20M,nodev,uid=1000,mode=0777,gid=4, 0 0 

This way /var/log is owned by volumio and everyone can write. Maybe there are better ways to do this, open for suggestion..
But this just works

gkkpch, 2015-11-16:
Using this on Odroid does not seem to make make any difference. 
After booting, the permissions are still 755 and ownership is *root*.
Is gid 4 supposed to be the 'adm' group?

volumio, 2015-11-16:
Yes, 4 is adm.
Is /var/log/ mounted on tmpfs ?

gkkpch, 2015-11-16:
yes, but when you make a typo, it won't work :) 
It's correct in the build master.

closed 2015-11-16 by gkkpch

volumio, 2015-11-16:
lol, glad it worked... still, if you have ideas on how to improved, they are very welcome!