volumio/Volumio2 #92: How to make it work ?

balbuze, 2015-11-16:
After installing Volumio0.92-2015-10-30PI.img.zip from repo, it works. Then I have updated like that 
- cd /volumio
- git pull
and for the web UI
- cd /
- wget http://repo.volumio.org/Volumio2/volumio-ui.tar.gz
- tar xv volumio-ui.tar.gz
and killall node
After that node stop running.
When launching by hand
info: -------------------------------------------
info: -----            Volumio2              ----
info: -------------------------------------------
info: -----          System startup          ----
info: -------------------------------------------
info: Loading plugins from folder /volumio/app/plugins/
info: Loading plugins from folder /data/plugins/
info: Loading plugin "system"...
info: Loading plugin "network"...
info: Loading plugin "networkfs"...
info: Loading plugin "services"...
info: Loading plugin "volumiodiscovery"...
*** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
*** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
info: Started advertising...
info: Loading plugin "alsa_controller"...
info: Loading plugin "upnp"...
info: [1447702084212] Starting Upmpd Daemon
info: Loading plugin "alarm-clock"...
info: Loading plugin "albumart"...
info: Plugin airplay_emulation is not enabled
info: Loading plugin "dirble"...
info: Plugin example_plugin is not enabled
info: Loading plugin "mpd"...
info: Loading plugin "spop"...
info: [1447702087047] ControllerSpop::loadTracklistFromDB
info: Loading Spop tracklist from DB...
info: Plugin mpdemulation is not enabled
info: Loading plugin "websocket"...
info: BOOT COMPLETED
Express server listening on port 3000
info: [1447702090841] CoreStateMachine::resetVolumioState
info: [1447702090849] CoreStateMachine::getcurrentVolume
info: CoreCommandRouter::volumioRetrievevolume
info: [1447702090896] CoreStateMachine::updateTrackBlock
info: [1447702090900] CorePlayQueue::getTrackBlock
info: SpopD Daemon Started
info: [1447702091059] Upmpcli error: Error: Command failed: /bin/sh -c upmpdcli -c /volumio/app/plugins/audio_interfaces/upnp/upmpdcli.conf -f 'volumio2'
src/upmpd.cxx:363::Can't open pidfile: flock failed. Return (other pid?): 740

info: [1447702091082] ControllerSpop::pushError(Error: Error reading DB: NotFoundError: Key not found in database [tracklist])
MPD error: Error: connect ECONNREFUSED
events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: connect ECONNREFUSED
    at exports._errnoException (util.js:746:11)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1010:19)

I had to configure spop or it stop with a error
What am I doing wrong ?

volumio, 2015-11-16:
You simply had the old volumio instance running... A reboot will solve everything

remember to give proper permissions to /volumio

balbuze, 2015-11-17:
you are right, mpd is not running.. If type mpd --stderr, I get \:
mpd --stderr
db: Database corrupted
fatal_error: block without 'plugin' name in line 31

I have re-installed mpd but nothing better.
The audio device is the onboard card. Alsamixer shows a  volume control.

balbuze, 2015-11-17:
So, I removed /volumio ( rm -Rf /volumio) reinstall it and remove credential from spopd.conf and reboot.
Now the end of the message is
info: BOOT COMPLETED
Express server listening on port 3000
info: [1447791657578] CoreStateMachine::resetVolumioState
info: [1447791657587] CoreStateMachine::getcurrentVolume
info: CoreCommandRouter::volumioRetrievevolume
info: [1447791657642] CoreStateMachine::updateTrackBlock
info: [1447791657647] CorePlayQueue::getTrackBlock
info: SpopD Daemon Started
info: [1447791657812] Upmpcli error: Error: Command failed: /bin/sh -c upmpdcli -c /volumio/app/plugins/audio_interfaces/upnp/upmpdcli.conf -f 'volumio2'
src/upmpd.cxx:363::Can't open pidfile: flock failed. Return (other pid?): 881

info: [1447791657837] ControllerSpop::pushError(Error: Error reading DB: NotFoundError: Key not found in database [tracklist])
MPD error: Error: connect ECONNREFUSED
SPOP command error:
{ [Error: connect ECONNREFUSED]
  code: 'ECONNREFUSED',
  errno: 'ECONNREFUSED',
  syscall: 'connect' }
SPOP status error:
{ [Error: connect ECONNREFUSED]
  code: 'ECONNREFUSED',
  errno: 'ECONNREFUSED',
  syscall: 'connect' }
events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: connect ECONNREFUSED
    at exports._errnoException (util.js:746:11)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1010:19)

of course there is still this problem with mpd...

mpasteur, 2015-11-18:
your's mpd problem is due to an error in mpd.conf file.
At line 31 you have an empty input{} block. If you comment it it should be fine.

closed 2016-01-17 by volumio