volumio/Volumio2 #94: nfs mount

balbuze, 2015-11-18:
To be able to mount a nfs filesystem, nfs-common is required. (apt-get install nfs-common).
But the system still try to mount a cifs fs even if the file "/data/configuration/system_controller/networkfs/config.json" contains "fstype": {
        "type": "string",
        "value": "nfs"
      },
"

gkkpch, 2015-11-21:
can confirm, nfs mount does not work, it defaults to cifs.
The nfs-common issue has already been addressed by adding it to the build scripts.

balbuze, 2015-11-21:
I can mount by hand after installing nfs-common. Then the mount appears in the webUI

gkkpch, 2015-11-21:
Hi,
yes, that is why nfs-common has been added to the 'recipes' which is used to run the first part of the volumio build script. This part more or less takes care of building the debian rootfs, including all common volumio parts, common to all other platforms (I think only PI and C1 in this case).
Are you using the build scripts yet?
In case you do, nfs-common should be covered.
Gé

balbuze, 2015-11-21:
I tried to build a image with build script, but with no success... Lack of instruction for me.

closed 2016-01-27 by balbuze