volumio/Volumio2 #97: new UI for monting a NAS volume

gkkpch, 2015-11-20:
I tried the latest UI for mounting a NAS volume and have to issues:
- there are 3 fields: ID, NAME and PATH. I noticed you have to put the share name in ID (??) and leave the other 2 empty. Probably the mapping is not correct yet. 
- my share has two top directories, only one of them is being indexed. The 2nd one is called **My Music (Gé)** (note the special character), on the mounted directory it shows as **My Music (G?)**. MPC won't index unless I rename it to **My Music (Ge)**. With the Volumio 1.55 configuration, special characters are not an issue. Could this a setting problem in mpd?

balbuze, 2015-11-21:
hi!
Have you tried to add plenty of songs to the queue by selecting a folder containing 2000 songs? I find it is slow. And it slow to swap beteween pages browse / playback /queue.

gkkpch, 2015-11-21:
Yes, I did. 687 folders, 9335 files took a little less than 10 minutes. Approx. half of it is .wav, the rest is .flac. Only the .flac files are tagged. I'm using a C1+ (1.5Ghz 4-core Cortex A5)
I can time the 2nd folder for you once the charset issue has been resolved (1089 folders, 13673 files)

volumio, 2015-11-21:
Thanks for all the feedbacks. I think what we need to do:
- Solve the issue with characters (enable utf8 encoding?
- Increase indexing speed (guess it is due to mounting params

gkkpch, 2015-11-21:
Update
- removed the mount
- changed filesystem_charset in mpd.conf to ISO-8859-1 
- re-created the mount -> mpd now scans both top folders
  1'776 folders, 23'008 files in 23min 27secs, mount type cifs 
- removed the mount, changed back to UTF-8, re-created the mount -> only one folder gets scanned

volumio, 2015-11-21:
That's a great finding!

volumio, 2015-11-21:
https://github.com/volumio/Build/commit/5b91c108045b5497fbf26ca425d91e0ccb7ffc74
For ISO charset... Let's see

gkkpch, 2015-11-21:
When removing an existing, mounted network share, it does not get unmounted.
An mpc update does not happen yet either.

balbuze, 2015-11-21:
Careful to UMOUNT the folder before erasing it.... I did it ;-( and lose files the time to stop the command...

gkkpch, 2015-11-21:
Well, one should use a backup share out of principle, testing with live data is risky at least ;)

gkkpch, 2015-11-21:
updated first post for transparency

volumio, 2015-11-21:
What if we mount it with no write permission?

gkkpch, 2015-11-21:
doubt if that changes anything, still think in case you remove a share (likely to be mounted already), that it needs to be unmounted first, then the mount folder removed and then run mpc update to clean the library. More or less opposite of creating a share.

balbuze, 2015-11-22:
with filesystem_charset "ISO-8859-1" in mpd.conf,  I'm having problem displaying accented characters
in album name displayed in browse page, but correctly displayed when enter in the album

gkkpch, 2015-11-22:
was afraid of that, I think we need further clarification. Would be interesting to know, why UTF-8 was OK for Volumio 1.55 and not for 2.0 (some default OS setting??)

balbuze, 2015-11-22:
Locale are missing in v2. When apt-get something it complains. I can' t
test before tomorrow it help...

volumio, 2015-11-22:
I guess because no default locale is set...

gkkpch, 2015-11-23:
I can't confirm that behavior, here it works with e.g. Ali Farka Touré 
(btw. the title underneath is without é).

screenshot browse tab

balbuze, 2015-11-23:
I have installed locale (sudo apt-get install locales) and configure it . Until now,  no success... I'll try to switch mpd.conf to UTF-8 
Deleting a mount from the UI seems to still not work a expected.

gkkpch, 2015-11-23:
I found another way to get it working without changing mpd.conf
Would be nice if balbuze could test to see if there is still a problem now.

In /volumio/app/plugins/system_controller/networkfs/index.js I changed line 172 to:
```javascript
mountutil.mount(pointer,mountpoint, { "createDir": true,"fstype": "cifs","fsopts":credentials + ",ro,iocharset=utf8,dir_mode=0777,file_mode=0666"}, function(result)
```

balbuze, 2015-11-23:
Now, it works for me by resetting mpd.conf with UTF-8 and using locale = UTF-8
I'm going to test your solution. I wonder if i need to rebuild the mpd db each time or changes will take in charge ?

balbuze, 2015-11-23:
In volumio 1.55, locale is set to en_GB.UTF-8

gkkpch, 2015-11-23:
Hmm, I made a new share with special character in folder and filenames, just a small number of files,  and deleted the share first when I was testing a different mount option. Only to make sure, that I can compare

volumio, 2015-11-23:
Great solution, which locale do you think we should use then?

gkkpch, 2015-11-23:
I don't think the locale has any influence on this problem. With the utf8 option in the mount, the filenames should be translated correctly. Could it be, that Volumio1.55  did just that?

This works when CONFIG_NLS_UTF8 is set in the kernel.

volumio, 2015-11-23:
So, without further actions, enabling iocharset=utf8 on the mount options does the trick?

gkkpch, 2015-11-23:
Yes

volumio, 2015-11-23:
Great I'll add it tomorrow...

volumio, 2015-11-24:
I'm trying to set up NFS mount mechanism, but it fails miserably every time... lol

What command do you guys use?

balbuze, 2015-11-24:
first you have to install nfs-common
I use the follwing command  "mount -t nfs 192.168.0.16:/mnt/HD/HD_a2/musique /mnt/NAS/nfsmount/"
But I don't manage access right....

volumio, 2015-11-24:
I do this as well, but miserably fail...

gkkpch, 2015-11-24:
This is how what works for me:
```javascript
mount -t nfs -o ro,rsize=8192,wsize=8192 192.168.123.250:/volume1/volumio-test /mnt/NAS/nfs-test
```
However, as permissions are granted on the nfs server, this requires proper mapping of users.
For this test I created 2 users, mpd and volumio (as I'm not quite sure, which one actually accesses the server) with permissions on 'volumio-test' and used squash option 'no mappings'.
There are probably more ways of doing this, you could e.g. map all users to the 'guest'-account.

balbuze, 2015-11-24:
what is the error message ?

volumio, 2015-11-24:
mount.nfs: access denied by server while mounting ...

gkkpch, 2015-11-24:
yes, that seems to be the problem, you need to have the correct permissions on the server, you can't influence that from the client. You can mount anything, but during mounting the permissions on the mount folder get set to what is allowed from the remote server.

gkkpch, 2015-11-24:
Can you list the contents of the server's /etc/exports?

gkkpch, 2016-01-07:
Similar to #104, closing this one

closed 2016-01-07 by gkkpch