volumio/Volumio2 #100: main window: previous, play, stop, next

gkkpch, 2015-11-22:
The buttons at the top of the main window don't seem to be enabled yet.

mpasteur, 2015-11-23:
hi,
i know that this type of reponse doesn't really help but "it works for me".
i've last ui from repo and i'm uptodate with the github repo.
did you see the "Client requests Volumio play" and "Client requests Volumio stop" on the console ? It can help to track if this is an UI or backend bug.

gkkpch, 2015-11-23:
Update
After some more testing realised what is happening and can reproduce it easily
The controls definitely don't work when select a track and change to the Playback tab.
This is what I was doing. The controls do work when you select a track and either stay on the Browse tab or move to the Queue tab. Every works, until you switch to the Playback. This is where not only the controls stop working, the remove button on the Queue tab and the buttons on the Browse tab don't work any more either.

Can you confirm this?

mpasteur, 2015-11-23:
i can't reproduce it but maybe someelse could. When i select a folder (or
single track) from browse everything is fine (except a big latency) even on
playback tab...
does your console react to your clicks ?

gkkpch, 2015-11-23:
No, as soon as I change to the playback tab, console stops as well, until I refresh the url and get back to the home screen.

volumio, 2015-11-23:
I noticed there's a bug which crashes the WS interface... I am trying to fix it...

gkkpch, 2015-12-12:
fixed

closed 2015-12-12 by gkkpch