volumio/Volumio2 #101: Browse does not display all tracks & titles

gkkpch, 2015-11-23:
My library holds ripped copies of all my CDs in .wav format. Except for the album name, the artist, tracknumber ans trackname, there is no tagging.

My problem is that the folders are displayed incompletely when I browse through them.
Sometimes I see a few, most of the time only one or two track/album title lines.
The remaining tracks are listed but the lines are empty.
Example.
with mpc listall I get for this album:
NAS/volumio-test/CD - HQ Music/Jimmy Nail - Growing Up In Public/01 - Ain't No Doubt.wav
NAS/volumio-test/CD - HQ Music/Jimmy Nail - Growing Up In Public/02 - Reach Out.wav
NAS/volumio-test/CD - HQ Music/Jimmy Nail - Growing Up In Public/03 - Laura.wav
NAS/volumio-test/CD - HQ Music/Jimmy Nail - Growing Up In Public/04 - Waiting For the Sunshine.wav
NAS/volumio-test/CD - HQ Music/Jimmy Nail - Growing Up In Public/05 - Real Love.wav
NAS/volumio-test/CD - HQ Music/Jimmy Nail - Growing Up In Public/06 - Only Love (Can Bring Us Home).wav
NAS/volumio-test/CD - HQ Music/Jimmy Nail - Growing Up In Public/07 - Wicked World.wav
NAS/volumio-test/CD - HQ Music/Jimmy Nail - Growing Up In Public/08 - Beautiful.wav
NAS/volumio-test/CD - HQ Music/Jimmy Nail - Growing Up In Public/09 - I Believed.wav
NAS/volumio-test/CD - HQ Music/Jimmy Nail - Growing Up In Public/10 - Absent Friends.wav

When I browse I get something like this :
 
[image]

volumio, 2015-11-24:
Interesting.. Can you try to browse em with gmpc for example?
So we understand if the problem is in mpd or volumio

gkkpch, 2015-11-24:
Yes, I will do that tonight (should have thought about that myself)

----- Reply message -----

&mdash;

gkkpch, 2015-11-24:
OK, this seems to be an mpd issue indeed, gmpc only shows the ones from the screen-shot above and leaves the rest out, that is even worse. 
I quickly checked Volumio 1.55, but this displays the tracks with their complete file name.

volumio, 2015-11-24:
That is a clue indeed, I'll update mpd

gkkpch, 2016-01-09:
Any news on this one?
Volumio 1.55 at least shows the filename, a must for people like me that have their whole CD library ripped on a NAS.

gkkpch, 2016-01-21:
Any news on this one?

gkkpch, 2016-02-10:
Update: I believe, song titles in .wav albums should be taken from the .wav-file basename
(eg. "01 - sing me a song.wav" should give "01 - sing me a song". Currently it seems totally random, most of the titles are shown empty, obvious because "normal" .wav files do not have embedded tags.

volumio, 2016-02-10:
I thought we had fixed this, but seems no.. We need to put 
if (title == undefined) show filename... in mpd controller

gkkpch, 2016-02-10:
yes! preferably without the ".wav" extension

gkkpch, 2016-02-25:
Getting there !
Browsing is now OK, but the Queue Tab still isn't (not showing all file titles)
Cosmetic: the Playback Tab could do without ".wav" in the tittle.

This is starting to look great, thank you!

gkkpch, 2016-02-25:
Comparing Volumio 1.55 Playlist and Volumio 2 Queue:
Volumio 1.55:
[image]
Same music folder with Volumio 2 Queue, lot's of titles missing:
[image]

apinner, 2016-04-03:
Is this still an issue in RC1 @gkkpch?

gkkpch, 2016-04-03:
Yes, it obviously will not have the highest priority (it plays) but would be nice when solved. 80% of my valued library consist of my ripped cd collection...

apinner, 2016-04-04:
So just so I'm clear, this is because your wav files have missing tags? In 1.55 it would just display the file name instead?

gkkpch, 2016-04-04:
Exactly, my CD's are ripped without adding tags as additional data chunks (which I believe is not a standard, though wave and aiff would support it).
The title would only have to be the base filename, the extension could be omitted.
It looks as if it was intended to be that way originally, as in some instances, it does list the filename, album and artist (see first screenshot above). I can give more samples per PM if you like.

apinner, 2016-04-04:
@volumio looks like you already know how to fix this one?

apinner, 2016-04-12:
Fixed in https://github.com/volumio/Volumio2/commit/93f26f2c1af9d2a4a4446371843dd762c995a5f2?

gkkpch, 2016-04-12:
I will check this tonight

gkkpch, 2016-04-12:
Now it is at least as good as Volumio 1.55 :)
Thank you guys!

closed 2016-04-12 by gkkpch