volumio/Volumio2 #102: Fresh Install of git repo

learntofly83, 2015-11-24:
Currently the command"
git clone https://github.com/volumio/Volumio2.git volumio
does not clone the http folder (im not a git expert)

Fix is to either use --recursive
git clone --recursive https://github.com/volumio/Volumio2.git volumio

Or go back to a manual SVN checkout
"preinstall": "svn checkout https://github.com/volumio/Volumio2-UI http/www",

jomo, 2015-11-24:
Hmm, d8bcb8cb0743a6b1a7333d950ddc17f28caa9ce6 introduced running `git submodule init && git submodule update` automatically via `npm install`.
Then for some reason @bbb42a7505310724d4601f070c1eb07e911ec996 switched that back to svn, but left the `www` git submodule.
Finally, @efbb6b33026c3c473699afc4dbd80c5573e50ae3 removed that as well, but the commit message doesn't contain any description as to why that was removed.

@volumio can you shed some light?

learntofly83, 2015-11-24:
Apologies if I have not followed the correct protocol of reporting this, Im not a programmer, more of a tinkerer.. The fresh alpha 4 web interface didnt work for me out of the box, on debian unstable so i "rm -R /volumio" and have been scratching my head for hours why recloning the repo wasnt bringing up the web UI...

volumio, 2015-11-24:
Yes, you guys are right that is not clear at all. We're still looking for the best method to have the UI always updated, and available both for cloning and in build scripts.
That's why now we have it tarred at http://repo.volumio.org/Volumio2/volumio-ui.tar.gz

We're looking to have the time to build a Travis CI hook and keep it on the branch... Still WIP then...

jomo, 2015-11-24:
Here's how that could work: 
- Every time you have a stable build of the UI, [follow the README](https://github.com/volumio/Volumio2-UI/blob/4fcbbdfb822ee89c2c7a4e74d5f52f34a6fed521/README.md) and build to the `dist` branch
  - This is currently done manually
  - could be automated with travis hooks, as you mentioned
- The `preinstall` script could checkout the latest version automatically.

Example:
```js
"preinstall": "git submodule init && git submodule update && cd http/www && git checkout origin/dist"
```

volumio, 2015-11-24:
One of the reasons is that we found submodule to be quite unconfortable... other ideas?

jomo, 2015-11-24:
I think submodules (being part of git, after all) is superior to svn checkout inside a git repo or downloading and extracting a tar.gz archive, especially when `npm install` does all of it without the need to run any of the submodule commands.
I don't think there is a proper different way, really.

The only other option I can think of is keeping the UI in the same repo and building it into the `http/www` folder.

We could also put the submodule update commands in a `update_ui.sh` script so it can be ran both manually and automatically via `preinstall`.

jhoutman, 2015-11-24:
Submodules work horribly for some clients there are out for OSX. But using SVN is worse :D. I would suggest NPM and run watchers in the development environment when upping the machine. Tricky part is to let Gulp not break, but there are some tricks with that.

I also prefer using a package manager rather than submodules. Even more when it's being updated on a regular basis.

volumio, 2015-11-24:
The problem is that we're not going to use npm install in the build script... It must run inside a qemu environment, and it simply dies when he needs to compile some native modules...

Probably our best options is to have a repo somewhere for modules (in ARM and X86 version, ideally built by a CI system) and UI (which doesn't need to differentiate between arm and X86).
Then the install script will just clone both (modules and UI) in the appropriate folder...

jomo, 2015-11-24:
> Submodules work horribly for some clients there are out for OSX

Not sure if I understand you correctly here, but if you don't use _the_ git client you're doing something wrong. I'm on OS X and have no problems at all.
Anyway, the npm script calls `git` which is (obviously) expected to be a git-compatible git client, you'd also expect `bash` to be bash-compatible.

If you don't use npm, how do you manage all the dependencies?
And in that case it should be fine to include the submodule update script in the npm package.json to allow people (such as @learntofly83) to easily install the repo?

volumio, 2015-11-24:
We of course use NPM for handling depencencies, the problem is that we can't call npm install when building, because building modules will make it crash.
So, we can keep npm install with its scripts for people cloning, but we must find a way to download UI and modules in build script..

jomo, 2015-11-24:
Alright, I suggest keeping the submodule & handle installation via npm and open another issue to find a solution for builds.

learntofly83, 2015-11-24:
As a side not in assume you are planning on using some iptables rules to route port 80 to port 3000 so node can be run as non root yet still accessed on the standard port?

volumio, 2015-11-24:
Already done ;)

learntofly83, 2015-11-24:
Well just wanted to say thanks to yourselves and everyone else o has contributed to this project.. My DAC has only an asynchronous USB input and I was concocting very expensive solutions to get it working with an Apple TV / airport express / Chinese clone whilst keeping it all digital.

I have 5 RPI's in the house so you've literally made my day (I've gone back to the stock 1.5x build)

Ps the alpha build count really benefit from a recent copy of raspi-config in order to make overclocking the PI2 easy 

Thanks again

volumio, 2015-11-24:
Yes, we'll include it! 
And thanks for your contributions, you're really welcome if you want to join us developing it!

closed 2016-01-17 by volumio