volumio/Volumio2 #104: NasMounts config.json missing 'Path' variable

ecto1a2003, 2015-11-26:
the 'Name' is used as a path leading to mounts that are multiple folders deep. for example 'e/music' is the path to the share but the mount folder is created with the same path

balbuze, 2015-11-26:
yep it's a known problem.  Name is used as path... you can edit config.json to correct it, waiting for a fix.

volumio, 2015-11-27:
Yep, I accept suggestion on how the input fields should be named, and how many of them ...

balbuze, 2015-12-30:
1 - There is still a problem mounting nfs share : name is used as path. 
2 -And it is not possible to mount the same folder once with cifs, once with nfs (to compare cpu use because cifs seems use up to 3% with a rpi2) because the mount path cannot be define properly ( trough webUI)

mpasteur, 2016-01-08:
For me we should have : 
 * an host (IP adress or hostname) eg : 192.168.0.10 or mynas
 * a path which is relative to the host : mp3/classical
 * a name - fixed by user to easily find their shares : NAS1, NAS2 ...
 * maybe a login /password 

did i miss something ? i can work on this this wek-end i think.. (mostly with NFS i don't know if i can mount  a cifs share to test)

balbuze, 2016-01-12:
With the last built (from today 12/01/16) I think that network drive configuration is not clear. With cifs files, it's rather easy even if the name of the share and the path should be separate. With nfs same thing but path is much more complicated.

balbuze, 2016-01-12:
Mpasteur I read your comment after I post...

mpasteur, 2016-01-13:
no problem. If you have some time to test my PR i would be glad to have some feedbacks...

balbuze, 2016-01-13:
I'll try tonight. But I see a lot of activities !

balbuze, 2016-01-13:
@mpasteur 
There is something I don't understand, with your last PR I see no difference. Maybe I miss something but I thought you add field to configure as you wrote above

volumio, 2016-01-13:
You need to update both the UI and the Backend, plus you need to delete /data/configuration and reboot

balbuze, 2016-01-15:
Hello,
I have tested the PR from mpasteur. For me, it's good, understandable settings etc. Some minor improvements : window is not refreshed when adding or removing share. We should disallow the use of space in name as it use to create the mount point and does not work.

ecto1a2003, 2016-01-17:
i think im running the latest build and i dont see an option in the ui for path, how do i update both the backend and the ui.

ecto1a2003, 2016-01-17:
... and it just magically fixed itself. weird?

balbuze, 2016-01-17:
@ecto1a2003 you'll find some useful information here : https://gitter.im/volumio/Volumio2

closed 2016-01-17 by volumio