volumio/Volumio2 #106: Sounds Terrible

ecto1a2003, 2015-11-27:
im running on a standard raspberry pi 2 and it sounds truly terrible no matter what i play or listen with. The previous version ran fine albeit that was through a hifiberry.

jomo, 2015-11-27:
Are you sure this is a problem with Volumio2 and not with your RasPi or setup?

balbuze, 2015-11-27:
If I understand, you use the onboard sound card to test V2?
If yes, you can't  compare the onboard output with a external DAC ( hifiberry in this case). The gap is very important.

ecto1a2003, 2015-11-27:
Yes, I am using the internal card as the hifiberry is nonfunctional at this time. Even still the sound I'm getting is atrocious and sounds more like interference than the song in listening to. 1.55 sounded amazing but I was using the hifiberry at the time.

tomatpasser, 2015-12-03:
The onboard output is terrible on the pi. What you describe sounds exactly like my experiences with onboard audio output. Try using volumio 1 without your hifiberry to verify.

ecto1a2003, 2015-12-03:
I did a pull yesterday and everything sounds fine, maybe it was a connection issue, maybe not. This issue is resolved for me.

closed 2015-12-03 by ecto1a2003