volumio/Volumio2 #109: Beta boot errors on RaspberryPi B

LouisKottmann, 2015-11-27:
Hello,

I just installed the "beta1" on my RapsberryPi B, and it does not boot.

So I plugged a monitor on it via HDMI to see the boot process, and it got stuck when trying to find the third partition (mmcblk0p3 - data).

It also noted that it was trying to make the partition, only the commands to do that were broken.

Ok no problem, I take back the card and add a new ext4 partition called "data" via gparted.

Back into the RPi, boot up, it finishes initramfs and I get a new error:

```
Kernel panic - not syncing: Attempted to kill init! errorcode=0x00000004
```

Here's how it looks:

[image]

Hope this helps you!

volumio, 2015-11-27:
Yes, it helps. The problem here is that the PI tries to load the V7 kernel, while it should load the V6. We're trying to understand why it's happening

atze09, 2015-11-29:
Same here.
Tried with 4GB and 8GB sd card, should i try 16GB?

LouisKottmann, 2015-11-29:
The problem does not come from the size of the card, as the image is about 3.5GB, it fits in a 4GB sd card.

Apparently, once you added the "data" partition, the problem is that the boot process tries to load a kernel for ARMv7 processors, and RaspberryPis (except the v2 and Zero versions) have ARMv6 procesors. That makes the kernel panic (stop everything).

Volumio devs have to resolve that issue and release a new beta version and unless you can help, you have to wait ;)

gkkpch, 2015-11-30:
the (wrong) kernel is already loaded when initramfs starts, initramfs itself runs correctly upto very the last statement where "exec switch_root" is done to start the init process with the prepared rootfs. This is where the kernel panic occurs.

volumio, 2015-11-30:
Thanks for the hint, we're looking at it and seems that something is actually wrong in this exact phase

gkkpch, 2015-11-30:
also check, that the initramfs is not too big and overwrites memory where it shouldn't. This happened when I had my go at the initramfs for the Odroid. You might have to adapt /etc/initramfs-tools/initramfs.conf from MODULES=most to MODULES=dep.

dominikhura, 2015-12-05:
Hello,

For me with Raspberry B+ I I see a few Times Illegal instruction. 

This is perhaps this ARMv7 version problem. Any ideas? Do you have any plan when to solve this issue?

gkkpch, 2015-12-05:
The PI maintainers are investigating the issue. If it was something obvious, the solution would have been there already. Let's give them some more time...

volumio, 2015-12-16:
It's proving to be challenging...

jncraig, 2015-12-19:
I can't get BETA1 to load on my new Pi B or on an old B+.  Both actually start to load but both fail.

The Pi B ends up in Maintenace Mode.  What kind of debugging information would be useful to you?

gkkpch, 2015-12-27:
I'm pleased that the root cause has been found and the problem solved. A PR is following.

volumio, 2015-12-27:
Well done gé.  I ll release a new beta asap

LouisKottmann, 2016-01-07:
@gkkpch @volumio any news on this? Do you need help?

gkkpch, 2016-01-07:
Hi Louis,
thanks, the problem has been solved (see above), including auto-resize the sd card. 
A new beta will be uploaded soon.
This might take a few days more because a series of other important updates need to be included.
If you want to build the image in the meantime: the build scripts are all up-to-date.

LouisKottmann, 2016-01-08:
I see, well I can wait a few days :)

Thanks

Hammering, 2016-01-15:
This was fixed at least for me in Alpha4 (2016-15-01). I don't know if it matters, but I run it with an 8Gig SD-Card.

volumio, 2016-01-17:
Fixed! Great work gé!

closed 2016-01-17 by volumio