volumio/Volumio2 #126: Fresh install

Domin6, 2015-12-26:
Hi Volumio team

This is a description of my attempt to get volumio to compile.

I did a fresh install of Debian 8.2.0, created a volumio user and the followed the guide.

I found a few missing packs that you need before you are able to follow the guide, i will list them here:

build-essential
curl
git

I then had to run the following to get node and npm installed

curl -sL https://deb.nodesource.com/setup_0.12 | sudo bash -
sudo apt-get install -y nodejs

(remember to check the script before blindly running code with sudo)

Now i was able to start nmp install which seem to finish ok, but when running node index.js i get the following:

info: Loading plugin "volumiodiscovery"...
*** WARNING *** The program 'nodejs' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=nodejs>
*** WARNING *** The program 'nodejs' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=nodejs&f=DNSServiceRegister>

It then continues on, until it finally crashes at this point:

info: BOOT COMPLETED
Express server listening on port 3000
events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: spawn upmpdcli ENOENT
    at exports._errnoException (util.js:746:11)
    at Process.ChildProcess._handle.onexit (child_process.js:1054:32)
    at child_process.js:1145:20
    at process._tickCallback (node.js:355:11)
    at Function.Module.runMain (module.js:503:11)
    at startup (node.js:129:16)
    at node.js:814:3

node and npm version is:
node = v0.12.9
npm = 2.14.9

Any help is appreciated, and perhaps the missing packages should be added to the guide.

Best regards
domin6

TermeHansen, 2015-12-26:
I can't remember where I found the hint, but I can see that I upgraded the upmpcli of debian to latest version?
http://www.lesbonscomptes.com/upmpdcli/downloads.html#debian

jyviko, 2015-12-26:
Hi Volumio team,
Had the same issue on Rpi2 running latest xbian. I upgraded to upmpcli as @TermeHansen suggested and got the following:
nfo: Loading plugin "websocket"...
info: BOOT COMPLETED
Express server listening on port 3000
info: [1451163627131] CoreStateMachine::resetVolumioState
info: [1451163627139] CoreStateMachine::getcurrentVolume
info: CoreCommandRouter::volumioRetrievevolume
info: [1451163627199] CoreStateMachine::updateTrackBlock
info: [1451163627204] CorePlayQueue::getTrackBlock
events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: spawn amixer ENOENT
    at exports._errnoException (util.js:746:11)
    at Process.ChildProcess._handle.onexit (child_process.js:1054:32)
    at child_process.js:1145:20
    at process._tickCallback (node.js:355:11)
    at Function.Module.runMain (module.js:503:11)
    at startup (node.js:129:16)
    at node.js:814:3

Any ideas? 
I would appreciate any help.

Best,
J

volumio, 2015-12-26:
Guys,  at this stage (and probably it will stay the same)  Volumio2  can only be installed on Volumio OS. There are lots of custom paths and permissions needed to make it work.  You can refer to Volumio Build repo for the build scripts and procedures...

Domin6, 2015-12-27:
Ahh thanks for clarifying this, perhaps it would be wise to make a note of this in the readme so you do not wast time answering questions like this ?

closed 2015-12-27 by Domin6

domi27, 2016-01-19:
Hi,

i have been also hit by this error even though i'm running on a Volumio2 VMWare instance (Beta1) and followed the "Preliminary Setup" instructions.

    info: BOOT COMPLETED
    Express server listening on port 3000
    events.js:85
          throw er; // Unhandled 'error' event
                ^
    Error: spawn upmpdcli ENOENT
        at exports._errnoException (util.js:746:11)
        at Process.ChildProcess._handle.onexit (child_process.js:1046:32)
        at child_process.js:1137:20
        at process._tickCallback (node.js:355:11)
        at Function.Module.runMain (module.js:503:11)
        at startup (node.js:129:16)
        at node.js:814:3