volumio/Volumio2 #127: Can't format an SD card correctly

gonace, 2015-12-28:
Hi 
I'm having problems getting the sd card partitioned correctly I've tried "ApplePi Baker", "Win32DiskImager" but the sd card get a "boot" partition of 60Mb another partition on 1.73Gb the rest is "free space".

When starting on the sd card no file or folder can be created due to lack of space, no file can be edited and so on.

So my question is: How do I format my 16Gb sd card, :(

volumio, 2015-12-28:
There is an issue currently with SD cards different than 8 gb. It will be solved by this weekend...  Thanks for reporting

gonace, 2015-12-28:
Oh
I see, thank you :)

gonace, 2015-12-28:
Is there a "fast" fix? I tried with fdisk but fdisk only recognized 3.5Gb of the 16Gb SD.

gkkpch, 2015-12-28:
When you have the 60Mb and 1.73Gb partitions, the card should be bootable. 
Upon boot, the initramfs should automatically create the volumio_data partition and grab the rest of the sd card for it.
Did you already try to boot?

gkkpch, 2015-12-28:
argh, that won't work, parted is used to find free space and will display >9999MB as GBs.
Now I get volumio's comment.

mpasteur, 2016-01-04:
Hi,
it's always the case if you write an image on your card the space availbale
is always the same as the image. It coult be disturbing if you have a "big"
card.
there is some tutorials to expand the / partition (difficult) or to add
another fs to store your files (easier)
some pointers (but search engines are your friends):
http://elinux.org/RPi_Resize_Flash_Partitions
http://raspberrypi.stackexchange.com/questions/499/how-can-i-resize-my-root-partition
...

cheers

luxus, 2016-01-04:
is this fixed by now?
where i can get the new img?

gkkpch, 2016-01-05:
Yes, it is fixed for those that can build the image. A new image will follow soon.

closed 2016-01-13 by volumio