volumio/Volumio2 #136: webradio

balbuze, 2016-01-06:
There is something wrong when sorting webradio by countries : for instance if you choose "France" it return nothing. Uk and Italy are ok. Didn't test them all

gkkpch, 2016-01-07:
Yes, this is strange. When I do an manuel 
"http://api.dirble.com/v2/countries/FR/stations?token=51cc13037f15d06e727b34fe02351caea896f357&page=1&per_page=30" in the browser, 
it returns me a list of french stations (the token is in the dirble config.json)
Have not checked it in detail, could it be a codepage thingy?

balbuze, 2016-01-07:
yes <i was thinking about a codepage issue. I'm trying to test it.

volumio, 2016-01-07:
I confirm I get the problem as well, thanks for spotting this... If you guys also could debug it, it would be fantastic...
I'm preparing the new build in the meanwhile... Optimizing a lot for old raspis...

balbuze, 2016-01-07:
With GB codepage, same problem..Don't know if it help, but if you select France, nothing happens, but if you wait you can hear the startup sound trying to play 1 sec and on and on

balbuze, 2016-01-07:
The only way to stop it is to killall node

gkkpch, 2016-01-07:
Ok, probably I can help from tomorrow night. Played around with dirble last weekend, the gathered debugging knowhow might help. 
There is another issue with lwebradio, like paging while listing statons. Currently you only get one page (the first) with a max. of 30 stations. This is not a bug, paging does not seem to be implemented yet. Though for a beta it might not be top priority, it does show the principle.

mpasteur, 2016-01-07:
datas returned by  dirble seems fine for code FR (cf attachment) so it should be an problem during "decoding" these datas. I'll try to debug it soon...
[dirble_FR.txt](https://github.com/volumio/Volumio2/files/81745/dirble_FR.txt)
[dirble_IT.txt](https://github.com/volumio/Volumio2/files/81750/dirble_IT.txt)

gkkpch, 2016-01-07:
Yes, already noted that (see 2nd post), any help in debugging appreciated.

Also noticed, that stations without streams (missing "streams" element) get listed, but obviously when you click them, they can't play. Suggestion: skip a station unless at least one stream url is specified.

mpasteur, 2016-01-07:
hmm maybe we don't work on the same version but on mine the bug is exactly here. Some stations don't have any stream so there is an exception [TypeError: Cannot read property 'stream' of undefined]. As there is no catch within this plugin the list doesn't show...

mpasteur, 2016-01-07:
if somebody could try this little patch..
It allow to display the FR country and UK...
It doesn't resolve the problem shown by gkkpch (https://github.com/volumio/Volumio2/issues/136#issuecomment-169809106) 

[patch_dirble_streams.txt](https://github.com/volumio/Volumio2/files/81809/patch_dirble_streams.txt)

volumio, 2016-01-08:
Wow guys! Could you send a PR?

mpasteur, 2016-01-08:
i've to clean my branch before that but yes i could

balbuze, 2016-01-08:
@mpasteur Your patch works great !

balbuze, 2016-01-08:
In the playback page, with some radio, no name for the radio (france inter), only text displayed mpd 44100khz 24bits

balbuze, 2016-01-08:
If I choose "france" I don't get same radio throught Volumio vs the website (  all "france-bleu xxx ) are not displayed in Volumio

gkkpch, 2016-01-08:
Have you tried "http://api.dirble.com/v2/countries/FR/stations?token=51cc13037f15d06e727b34fe02351caea896f357&page=1&per_page=30" in your browser? 
Also, you only get the first page, so there will be many stations missing

balbuze, 2016-01-08:
You're right : it's because it 's only the first page. I found what I was looking for page 20... Is there a way to get other page in Volumio ?

mpasteur, 2016-01-08:
balbuze : for your pb (https://github.com/volumio/Volumio2/issues/136#issuecomment-170040320) i've some difficulty to track this but it's seems t's because thestream itself doesn't return a name.
GET /11008/franceinter-midfi128.mp3 HTTP/1.1
User-Agent: Music Player Daemon 0.19.1
Host: audio.scdn.arkena.com
Accept: */*
Icy-Metadata: 1

HTTP/1.1 200 OK
Server: nginx/1.2.1
Date: Fri, 08 Jan 2016 22:04:17 GMT
Content-Type: audio/mpeg
Connection: close
Accept-Ranges: none
icy-br: 0
icy-genre: icecast
icy-name: franceinter-midfi128.mp3
icy-pub: 0
icy-url: http://www.icecast.org/
Cache-Control: no-cache, no-store
Pragma: no-cache
Access-Control-Allow-Origin: *
Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
Access-Control-Allow-Methods: GET, OPTIONS, HEAD
Expires: Mon, 26 Jul 1997 05:00:00 GMT
icy-metaint: 16000

==================================================
when there is a name playback screen can display it.
eg:
GET / HTTP/1.1
User-Agent: Music Player Daemon 0.19.1
Host: 80.13.146.243:8000
Accept: */*
Icy-Metadata: 1

ICY 200 OK
icy-notice1:<BR>This stream requires <a href="http://www.winamp.com/">Winamp</a><BR>
icy-notice2:SHOUTcast Distributed Network Audio Server/win32 v1.9.8<BR>
icy-name:ISABELLE FM  (91.3 - 102.9 Mhz)  FRANCE  ***  Dance - French Pop - Techno - House - Rap  ***
icy-genre:International
icy-url:http://www.isabellefm.com
content-type:audio/mpeg
icy-pub:1
icy-metaint:8192
icy-br:64

So it's seems that playback screen only display info available by mpd on the stream...

volumio, 2016-01-08:
Yes, to display other infos we need to change the webradios playback backend..

balbuze, 2016-01-13:
Add a radio to favourite radio does nothing

volumio, 2016-01-13:
Refresh the page and you'll see it. We must push it again after adding...

balbuze, 2016-01-13:
I restart from a new build ( from last script ) and sorry but I can't add radio in radio favourites.  No file is updated in /data/favourites

volumio, 2016-01-13:
Sorry I understand that you couldn't add to mywebradios... You're right , to fix

mpasteur, 2016-01-13:
i've found two sources for this problem
1 - addWebRadioToFavourites in UI just call addToFavourites but should emit
an addWebRadioToFavourites even on socket
2 - on the server side there is an lsinfo command which return :
{ navigation: { prev: { uri: 'http://uk1-pn.mixstream.net:8120' }, list: []
} }

and we try to get navigation.list[0].artist ==> Exception

i've to sleep i'll try to make a PR tomorrow

apinner, 2016-04-03:
@mpasteur did you manage to resolve this? I can't seem to find a PR.

apinner, 2016-06-27:
dirble replaced with shoutcast, no longer an issue

closed 2016-06-27 by apinner