volumio/Volumio2 #140: alsa_controller

gkkpch, 2016-01-07:
something is broken in the newest build, starting volumio.service throws the following error:
nfo: Loading plugin "alsa_controller"...
/volumio/app/plugins/audio_interfaces/alsa_controller/index.js:47
	self.commandRouter.sharedVars.registerCallback('alsa.outputdevice',self.outpu
	                              ^
TypeError: undefined is not a function
    at ControllerAlsa.onVolumioStart (/volumio/app/plugins/audio_interfaces/alsa_controller/index.js:47:32)
    at PluginManager.loadPlugin (/volumio/app/pluginmanager.js:81:28)
    at PluginManager.loadPlugins (/volumio/app/pluginmanager.js:154:22)
    at new CoreCommandRouter (/volumio/app/index.js:33:21)
    at Object.<anonymous> (/volumio/index.js:34:21)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Function.Module.runMain (module.js:501:10)

ecto1a2003, 2016-01-08:
i am also receiving this error, UI wont start

root@volumio:/volumio# node index.js
info: -------------------------------------------
info: -----            Volumio2              ----
info: -------------------------------------------
info: -----          System startup          ----
info: -------------------------------------------
info: Loading plugins from folder /volumio/app/plugins/
info: Loading plugins from folder /data/plugins/
info: Loading plugin "system"...
info: Loading plugin "network"...
info: Loading plugin "networkfs"...
info: Loading plugin "services"...
info: Loading plugin "volumiodiscovery"...
*** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
*** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
info: Started advertising...
info: Loading plugin "alsa_controller"...
/volumio/app/plugins/audio_interfaces/alsa_controller/index.js:47
        self.commandRouter.sharedVars.registerCallback('alsa.outputdevice',self.outpu
                                      ^
TypeError: undefined is not a function
    at ControllerAlsa.onVolumioStart (/volumio/app/plugins/audio_interfaces/alsa_controller/index.js:47:32)
    at PluginManager.loadPlugin (/volumio/app/pluginmanager.js:81:28)
    at PluginManager.loadPlugins (/volumio/app/pluginmanager.js:154:22)
    at new CoreCommandRouter (/volumio/app/index.js:33:21)
    at Object.<anonymous> (/volumio/index.js:34:21)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Function.Module.runMain (module.js:501:10)
root@volumio:/volumio#

gkkpch, 2016-01-09:
see #28 on volumio/Build

closed 2016-01-09 by gkkpch

ecto1a2003, 2016-01-13:
What do you mean by that, are we doing something wrong?. All I know is nothing works

gkkpch, 2016-01-13:
The issue was closed because a. it was the same as issue #28 in the Build repo and b. the issue was fixed in the meantime and tested.

ecto1a2003, 2016-01-15:
im still having the issue what do i need to do, i do a git pull origin master on /volumio what else?

volumio, 2016-01-15:
Yes, do a git pull on /volumio

ecto1a2003, 2016-01-15:
just did, exact same error.

volumio, 2016-01-15:
Ok, I'm doing a new build... 10 minutes and you'll have a fresh link...

volumio, 2016-01-15:
There you go
http://updates.volumio.org/pi/volumio/0.809/Volumio0.809-2016-01-15PI.img.zip

ecto1a2003, 2016-01-15:
excelent, that worked. thankyou.