volumio/Volumio2 #143: output setting

balbuze, 2016-01-09:
only the current output device is displayed : bcm2835 Alsa
If manually changed in /boot/config.txt (ex iqaudio-dacplus ) reboot and back to webUI, only this is displayed

gkkpch, 2016-01-09:
just to confirm this works fine on an Odroid C1, incl. USB Audio. There is a small glitch when hot-plugging the usb audio device: when you select this, save and then remove the USB device, it will show 'VALUE NOT FOUND BETWEEN SELECT OPTIONS' the next time you go to the playback options.

balbuze, 2016-01-09:
On RPI 2, when 2 output device exist, only the first (bcm2835) selected is used even when choosing the other one, save and reboot

Sceedster, 2016-01-22:
Sound output: I have a RPI2 + Hifiberry, please make it possible to choose my hardware. Oh, and thank you all for your hard work!

Sceedster, 2016-01-28:
adding the line to config.txt does not work....

apinner, 2016-04-03:
DACs introduced in RC1

closed 2016-04-03 by apinner