volumio/Volumio2 #150: Albumart suggestion

gkkpch, 2016-01-11:
I have a suggestion to improve display of albumart:

If embedded albumart can be found,  let that take first priority
Otherwise if a file called "folder.jpg" (exact name) exists, let that take 2nd priority (it is the fastest way plus the user put the folder.jpg there on purpose)
Otherwise if ".jpg" exists with the basename equal to the filename of the current song, use that as 3rd priority
Otherwise if a ".jpg" exists with the basename equal to the name of the current folder, display that as 4th priority
Otherwise if web albumart can be found, use that as 5th priority
In case any other image exists, pick one randomly as the last 

Reason being: the first 4 alternatives are deliberately put there by the user
Web albumart (Coveroart etc.) could be used if none of the first 4 exist, the 6th is just a last option.

Hammering, 2016-01-15:
If possible, I would like there be some sort of setting for the "default" .jpg name if this were to be implemented, that way those who already has a standard naming scheme doesn't have to change it. For example, in my library I have a standard where the album art is named "cover.jpg", when nothing is embedded. I know a lot use the "folder.jpg"- name, but there are other standards out there. 
I guess it would work with the "pick random image"- option as well as long as that comes before the webart, but that wouldn't always be ideal (say if you have for example the album art for the back of a cover as well). To force the 2nd-priority to a specific name should in my opinion be optional.

volumio, 2016-01-15:
I'm looking on how to implement id3 tag, and I find Hammering a good idea!

jomo, 2016-01-15:
`folder.jpg` is common because Windows used that as the folder icon in the File Explorer, however it also downscaled it to crappy quality so people started using `cover.jpg`. Anyway, `front.jpg`, `back.jpg`, `albumart.jpg` and a lot of others are also common.

ID3 embedded should always go first, but for the rest maybe there could be an option that allows multiple names, ordered by priority? There's already [a list of file names](https://github.com/volumio/Volumio2/blob/f589e62bb0073c898ae6b27755e0b01b320a3060/app/plugins/miscellanea/albumart/albumart.js#L52-L54), however the file extension should not be included IMHO.
It also doesn't seem to be user configurable, which would fix @Hammering's issue.

gkkpch, 2016-01-15:
I fully agree that there a other alternatives to folder.jpg and they should be considered too. Being able to choose a default, as Hammering suggests, would be ultimate.

apinner, 2016-04-03:
Album art is now working, closing unless there are any objections?

closed 2016-04-03 by apinner