volumio/Volumio2 #159: Send audio from main computer (pulseaudio) to Volumio

LouisKottmann, 2016-01-16:
Hello,

I got Volumio (Alpha4) to run and play sounds through a soldered hifiberry digi, NAS & radios working, great.
The last thing I need is to be able to send sound from my main computer running the latest Ubuntu (and thus, pulseaudio), to Volumio. Say, to watch movies with VLC.

Since Volumio does not seem to use pulseaudio, I can't simply add a "sink", so what would be the preferred way?

Regards,
Louis

ecto1a2003, 2016-01-18:
how did you get the hifiberry to work? mine still isn't?

LouisKottmann, 2016-01-18:
The web UI does not work, but if you follow [the instructions](https://www.hifiberry.com/guides/configuring-linux-3-18-x/) from hifyberry's website and then use the web UI, it works.

ecto1a2003, 2016-01-18:
cool, ill give that a try

ecto1a2003, 2016-01-19:
that worked a treat, thanks.

LouisKottmann, 2016-01-19:
Ok I did it!

My solution is to install pulseaudio on the Volumio machine. Here is how to do it:

```bash
ssh volumio@volumio.local
sudo -s

# Install pulseaudio
apt-get update
apt-get install libasound2-plugins \
                pulseaudio pulseaudio-utils \
                pulseaudio-module-zeroconf \
                pulseaudio-esound-compat

# Set pulse daemon config
cat <<EOF > /etc/default/pulseaudio
PULSEAUDIO_SYSTEM_START=1
DISALLOW_MODULE_LOADING=0
EOF

# Make local sinks available on LAN
load-module module-esound-protocol-tcp auth-ip-acl=127.0.0.1;192.168.0.0/16
load-module module-native-protocol-tcp auth-ip-acl=127.0.0.1;192.168.0.0/16
load-module module-zeroconf-publish

# Start pulseaudio
sudo pulseaudio --system --daemon
```

And then I had the audio sinks from the Pi available on my local machine :tada: 

@volumio this might desserve a mention somewhere in a wiki

balbuze, 2016-01-22:
Hi!
you can try this : https://volumio.org/forum/streaming-from-ubuntu-and-other-linux-volumio-t2914.html
You have nothing to be installed on your volumio device

LouisKottmann, 2016-01-22:
Hello!

That solution is not satisfying. 

First of all I tried it and it doesn't seem to work with Volumio alpha4-5 "out of the box". It was working during my PiMusicBox test though.

Even when it does, it actually re-encodes the sound, inducing very significant delay. We're talking multiple seconds. A direct pulse-to-pulse connection induces non-noticeable delays, making it a good way to stream the audio from movies.

I'm sure there's a way to make my solution work alongside Volumio's native Alsa, but I'm not expert enough in the .asound format!

jomo, 2016-01-22:
I think this would be a perfect 'output plugin', see #32.

closed 2016-07-10 by volumio

wvengen, 2016-11-30:
What would be the next step here? I'm not really sure whether pulseaudio could be a music service, or if it belongs more between ALSA and SOX or music services (or even pulseaudio could expose ALSA).

LouisKottmann, 2016-12-01:
Basically, all you have to do is be able to run:

```bash
sudo pulseaudio --system --daemon 
```

While not disrupting volumio use.

wvengen, 2016-12-03:
That actually works, thanks! Needed to edit `/etc/pulse/system.pa` to [enable networking](https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/Network/#index1h2), then together with [module-suspend-on-idle](https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/Modules/#index63h3) with a low timeout it's quite ok (I get a latency of 300ms). I also [included](https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=82839) the [tsched=0](https://volumio.org/forum/pulseaudio-network-sink-t773.html) option.

Remaining issues:

    I: [pulseaudio] (alsa-lib)utils.c: could not open configuration file /usr/share/alsa/ucm/IQaudIODAC/IQaudIODAC.conf
    I: [pulseaudio] (alsa-lib)parser.c: error: could not parse configuration for card IQaudIODAC
    I: [pulseaudio] (alsa-lib)main.c: error: failed to import IQaudIODAC use case configuration -2
    I: [pulseaudio] alsa-ucm.c: UCM not available for card IQaudIODAC

and

    I: [pulseaudio] alsa-sink.c: Driver does not support hardware volume control, falling back to software volume control.
    I: [pulseaudio] alsa-sink.c: Driver does not support hardware mute control, falling back to software mute control.

xipmix, 2016-12-08:
Could you post your system.pa for other people to try?

wvengen, 2016-12-09:
Sure thing: [system.pa](https://github.com/volumio/Volumio2/files/643363/system.pa.txt)

wvengen, 2016-12-12:
Regarding the message _Driver does not support hardware volume control_, it is caused by [`has_volume` being `false`](https://github.com/pulseaudio/pulseaudio/blob/8cda1fe3e2072aba21cfa9f90d707b029c07773c/src/modules/alsa/alsa-sink.c#L1427), which is [received from alsa](https://github.com/pulseaudio/pulseaudio/blob/8cda1fe3e2072aba21cfa9f90d707b029c07773c/src/modules/alsa/alsa-mixer.c#L1457) through [snd_mixer_selem_has_playback_volume](http://www.alsa-project.org/alsa-doc/alsa-lib/group___simple_mixer.html#ga17468934f36929ae2f7af4850f40f851).

Full pulseaudio verbose log: [volumio-iqaudioiodac-pulseaudio.log](https://github.com/volumio/Volumio2/files/647392/volumio-iqaudioiodac-pulseaudio.txt)
(Note that I've blacklisted `snd_bcm2835`, the internal audio card.)

wvengen, 2016-12-12:
Hmmm it _does_ look like UCM is important for finding the right mixing controls, as PA's [module-alsa-sink](https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/Modules/#index4h3) seems to suggest.

Check. When I disable `module-udev-detect` and add

    load-module module-alsa-sink control=Digital

hardware volume control works (as a consequence, pulseaudio's and MPD's volume control are now synced - convenient that I can now use the MPD control app to change the volume of pulseaudio streams as well).

I suppose the next step is writing a UCM file for IQaudIODAC.

wvengen, 2016-12-12:
related: https://volumio.org/forum/hardware-volume-control-iqaudio-dac-t1642.html

aiomaster, 2018-08-18:
mpd has a pulse output plugin, so in theory one could install the pulseaudio system wide and let mpd use pulseaudio as backend.
The problem is, that I could not test it, because the mpd version installed in the default volumio raspbian image (in my image it is 0.20.6) somehow has no pulse Output backend compiled in.
I was not able to build a new version of mpd at my pi, cause there are some modifications of volumio of some libs and I am not experienced enough to fix the details.

If the upstream mpd in debian could have the pulse output plugin compiled in, we could test to use pulseaudio and volumio together.