volumio/Volumio2 #164: Wireless

jncraig, 2016-01-18:
Alpha 4 is working for me, but I need some help with wireless.

Can someone please point me to some decent instructions on installing my usb wireless device?

Much appreciated!

purpleferret, 2016-01-18:
Hi jncraig,
you can configure wlan from the text console (keyboard and monitor connected).
Just edit two files like the following example (MYSSID is your wlan name and passphrase the password):
```
editor /etc/wpa_supplicant/wpa_supplicant_mywlan.conf
```
```
network={
    ssid="MYSSID"
    psk="passphrase"
}
```
```
editor /etc/network/interfaces
```
```
auto wlan0
iface wlan0 inet dhcp
    wpa-conf /etc/wpa_supplicant/wpa_supplicant_mywlan.conf
```
With this configuration it should connect automatically with every reboot or you can use ifup wlan0 to enable wlan right away.
If you need a more specific setup, have a look at the wpa_supplicant documentation.
Here is a nice HowTo documentation: https://wiki.debian.org/WiFi/HowToUse .

jncraig, 2016-01-19:
Thanks!

Your suggestions gets me part of the way there.

I can see wireless networks after making the configuration changes.  But, trying to connect to compufest-ap3 (or any other network for that matter) gives this error:

[image]

For what it's worth, the compufest-ap3 network has no security key -- simplifies testing.  After setting wpa_supplicant.conf to a minimal configuration

network={
    ssid="compufest-ap3"
}

it now looks like this:

ctrl_interface=/var/run/wpa_supplicant
network={
ssid="compufest-ap3"
key_mgmt = NONE
}network={
    ssid="compufest-ap3"
}

jncraig, 2016-01-19:
The first one is wpa_supplicant_mywlan.conf, the second is supplicant.conf.

volumio, 2016-01-19:
Please, wait for the official wi-fi conf to be released.

closed 2016-01-19 by volumio

jbmfg, 2016-02-12:
This still happening?

jncraig, 2016-02-12:
I don't know.  I've stayed away from playing with wi-fi until "official wi-fi conf" is released.  Has that happened?

jbmfg, 2016-03-17:
Still not working for me in latest beta.  Not sure what the deal is