volumio/Volumio2 #165: Comments and Observations

jncraig, 2016-01-18:
I've downloaded and installed the Alpha 4 image.  Here are some initial questions and comments.  I'm not quite far enough along to report bugs, as I suspect that most of these items have to do more with me than with Volumio 2.

I should point out that I've been using Volumio 1 with 2 systems, one with a HiFiBerry and one with a USB sound card, both with different WiFi dongles.  Both work fine.  So, I'm hoping to get these running with Volumio 2.  If there are things that don't exist yet, just let me know!

------------------------------------------------------------------------------------------------------------------------
UPDATE
The first think I did was click the Update button under System.  In no time, I was upgraded to "System version: 0.814 Released: Sun Jan 17 21:34:25 CET 2016."  Nice job!

------------------------------------------------------------------------------------------------------------------------
NAS
I have a WDMyCloudEX4 NAS with a public share that contains music files.  I can't seem to "mount" it.  In Windows, the share is \\wdmycloudex4\Public\Music-New, and it requires no username and password.

My settings are:
FS TYPE cifs
NAME me
PATH: \Public\Music-New
IP: 192.168.1.251 (Which is the IP address of the NAS.  

I couldn't figure out how to specify the full page, as it seem that entries were required in both PATH and IP.  

I didn't configure any other entries.

I get a "success" message and also one that says that the library scan will be done.

The status message is:
Name	Path	                                                Mounted	Size	
me	       192.168.1.251:\Public\Music-New	false	        26.88 GB

No music appears in the library.

-----------------------------------------------------------------------------------------------------------------------------
SOUND CARD
I have a USB sound card.  Should it work?  It does work for me with Volumio 1.  My only choice for playback is bcm2835 ALSA. 

Is there a way to install a USB Sound Card?

Similarly, I have a second system with a HiFiBerry board.  How would I install an i2s?

-----------------------------------------------------------------------------------------------------------------------------
WiFi
As I mentioned in another posting, I wasn't able to get WiFi working.  So, if there are tips for that I'd love to hear.

==============================================================================
In general, I've greatly impressed by Volumio, and I look forward to Volumio 2.  I'm eager to help in any way that I can (mostly testing, I assume).  Point me to the right place for that, too!

Thanks!

mpasteur, 2016-01-18:
Hi maybe I can help you with the mount problem. 
Your config seems good. Have you any error reported on log?  To retry you just have to edit the share and save it it will try to mount it again..

jncraig, 2016-01-18:
Thanks.

I did look at volumio.log and found some errors in the mount command.  I did some careful editing, and got past that part.  I'm not sure why what I posted was wrong, but the actual command seemed to have some errant \ characters.  I somehow got rid of them, mounted the NAS and it's indexing like crazy!

mpasteur, 2016-01-18:
I think you don't have to put the first \ in the path.  Will try it
tonight...

jncraig, 2016-01-19:
no leading and / rather than \ is what works for me.

atcapollo, 2016-02-07:
Wifi is using wep by default you need to edit the wifi conf at /etc/wpa_supplicant/wpa_supplicant.conf and add key_mgmt=WPA-PSK

apinner, 2016-04-03:
All should have been resolved in RC1

closed 2016-04-03 by apinner