volumio/Volumio2 #170: Web radio missing

ppizzo, 2016-01-18:
I'm using the latest version, either downloaded from the latest Alpha release or built by myself using the build script. In both cases the web radio list is missing: no web radio available. Either browsing by country or by genre. The output web page is empty.

I've checked DNS resolution and it works fine.

Is there a way to rebuild the list? I've noticed that there's no webradio share in Samba as version 1 and I'm wondering how to add my own web radio pls.

Thanks

volumio, 2016-01-18:
Hi, this is a temporary outage, since this evening Dirble's API system just put our API key on hold for too many requests...

closed 2016-01-18 by volumio

ppizzo, 2016-01-19:
Many thanks for your quick reply!