volumio/Volumio2 #171: DAC output setup?

ppizzo, 2016-01-19:
Hi, in Volumio2 seems to lack the *"I2S driver"* DAC setting which is available in Volumio 1. I've tried adding `dtoverlay=hifiberry-dacroot` option in `/boot/config.txt` but after a reboot still no music out of the RCA plugs.

Is there a way to enable DAC output on Volumio 2? I'm running a Raspberry Pi B (old model) with HiFiBerry DAC. The DAC works perfectly on Volumio 1.

volumio, 2016-01-19:
We still have to code it... is an early alpha...

closed 2016-01-19 by volumio

ppizzo, 2016-01-19:
OK, thanks. Anyway, after enabling the DAC in `/boot/config.txt` I selected the corresponding hw device in `/etc/mpd.conf`(identified from the output of aplay -l) and I was able to play music through the DAC! So I can confirm it works, it's just a matter of coding the web ui.