volumio/Volumio2 #179: volumio wont start cue-parser module missing

ecto1a2003, 2016-01-23:
info: Loading plugin "mpd"...
module.js:338
    throw err;
    ^
Error: Cannot find module 'cue-parser'
    at Function.Module._resolveFilename (module.js:336:15)
    at Function.Module._load (module.js:278:25)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (/volumio/app/plugins/music_services/mpd/index.js:15:14)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
volumio@volumio:/volumio$

balbuze, 2016-01-23:
I can't see this error. Have you done a "npm install" in /volumio ?

volumio, 2016-01-23:
Yep, I think you just updated the UI... while modules are stored here:
http://repo.volumio.org/Volumio2/node_modules_arm.tar.gz

closed 2016-01-23 by volumio