volumio/Volumio2 #188: RaspiDACRev.3 sound card support

jgrulich, 2016-01-25:
Please add the support for the raspidac3 overlay. If added to the config.txt, the card works, but the volume is set to 100% and can't be changed. The "Headphones" control works via amixer sset Headphones well. It will be fine if this control will be implemented in the Volumio too. Alsamixer already works with this.

apinner, 2016-04-04:
I can't read czech, but presumably this should just be a case of adding a new line in dacs.json?

http://www.grulich.eu/files/RaspiDACv3.pdf

volumio, 2016-04-04:
Yep!

apinner, 2016-04-04:
Added in https://github.com/volumio/Volumio2/commit/c08011ef497ef96464e279049168f5299424cac8

jgrulich, 2016-04-04:
There are two controls for this card. There is "Digital" for DAC volume control and "Headphones" for headphones amplifier volume control. If there is no option for second knob, than let's make two card versions, one with 100% "Headphones" and regulated "Digital" and second one swapped.

apinner, 2016-04-04:
`{"id":"raspidac3dac","name":"RaspiDACv3 DAC","overlay":"raspidac3","alsanum":"1","mixer":"Digital","modules":"","script":""},
{"id":"raspidac3hp","name":"RaspiDACv3 Headphones","overlay":"raspidac3","alsanum":"1","mixer":"Headphones","modules":"","script":""}`

???

jgrulich, 2016-04-04:
Yes, somehow it works. Exactly it regulates the given volume, but doesn't set the 100% for the second one. It lives it as it was before. Means that when I use the "DAC" and set the volume to 10% and than switch to the "Headphones" than I'm not able to increase the DAC to 100% until I use the alsamixer via the SSH.

jgrulich, 2016-04-05:
I've tried to use this:

{"id":"raspidac3dac","name":"RaspiDACv3 DAC","overlay":"raspidac3","alsanum":"1","mixer":"Digital","modules":"","script":"amixer -c 1 sset Headphones 40"},
    {"id":"raspidac3hp","name":"RaspiDACv3 Headphones","overlay":"raspidac3","alsanum":"1","mixer":"Headphones","modules":"","script":"amixer -c 1 sset Digital 200"}

but it doesn't work. Another way how to run the init script?

apinner, 2016-04-05:
@volumio any thoughts on this one?

jgrulich, 2016-04-05:
There is still the card auto-detection which works well and when detected "Card" it may be than selected the control "Headphones", or "Digital" in the part of "Volume Options" without reboot.
There is no need for the I2S DAC setup, but if needed, use only one line in the dacs.json:

{"id":"raspidac3","name":"RaspiDAC Rev.3","overlay":"raspidac3","alsanum":"1","mixer":"Digital","modules":"","script":""}

I've made the init setup for both controls in the rc.local:

amixer -c 1 sset Headphones 40
amixer -c 1 sset Digital 200

closed 2016-06-27 by apinner