volumio/Volumio2 #192: Browsing library

gkkpch, 2016-01-29:
current commit: browsing is noticeably faster than before!
Small issue: when selecting an album from the library, the queue shows more than one copy of the contents

balbuze, 2016-01-29:
for me I don't see that problem. In which case does it occurs ? With any album ? which file format (I use flac)

gkkpch, 2016-01-29:
most of them, some of them contain .wav, .flac, .mp3. I select one and then the songs in the playlist tab show up multiple times

balbuze, 2016-01-29:
I use the very last webUI (18e8a2252b0f8248301e6e3b219a81d018257dd6), and you ?

piercer, 2016-01-29:
I also see this issue. If you ad an album, then add another album you will see multiple entries for both albums in the queue. Flac format files.

gkkpch, 2016-02-01:
By coincidence used a .cue file today, this does not duplicate queue entries!

PR #367 (2016-04-20): Update fs-extra to version 0.28.0 🚀
Hello :wave:

:rocket::rocket::rocket:

[fs-extra](https://www.npmjs.com/package/fs-extra) just published its new version 0.28.0, which **is not covered by your current version range**.

If this pull request passes your tests you can publish your software with the latest version of fs-extra – otherwise use this branch to work on adaptions and fixes.

Happy fixing and merging :palm_tree:

---
The new version differs by 13 commits .

- [`ef35b70`](https://github.com/jprichardson/node-fs-extra/commit/ef35b7006ab302e2c6d20977a1e822d323681c6a) `0.28.0`
- [`8c56e5 [... cut]
closed 2016-02-11 by gkkpch