volumio/Volumio2 #218: reboot after update or bash cmd

balbuze, 2016-02-08:
After a update the system never reboot.
After a sudo reboot the system never reboot
Tested with RPIB and RPI 2

volumio, 2016-02-09:
thanks! I'll investigate why this happens..

muzbox, 2016-02-18:
Same here, after update this morning the system never restarted

muzbox, 2016-02-18:
I can log on via SSH but the website is not available

volumio, 2016-02-18:
But after a forced reboot (aka PSU removal) does the system restart fine?

muzbox, 2016-02-18:
No, I then went and downloaded the new beta. It loaded fine first time

The second I edit config.txt for my iqaudio-dacplus and reboot it stops working.
 Tried this a number of times.

balbuze, 2016-02-18:
In my case no problem after unplug - replug psu

volumio, 2016-02-18:
When you edit config.txt you must have messed it .
What did you put in it?

muzbox, 2016-02-18:
It is quite possible

initramfs volumio.initrd gpu_mem=16 force_turbo=1 dtoverlay=iqaudio-dacplus

is the line from the file

Interestingly though,I wonder why it worked before I upgraded and then not after without me editing the file.

This is my recent edit. So it is possible I have done something wrong. I could not find any documentation on how to do this correctly so if you can point out my mistake I would be grateful

muzbox, 2016-02-22:
hi

I have updated to the latest versions, the good news is it does not stop working when I add the overlay.

The bad news is the overlay does nothing. only ALSA showing in the playback, it is not picking up my DAC

muzbox, 2016-02-22:
this is my current version 

System version: 0.834 Released: Sun Feb 21 17:18:36 CET 2016

this is my config.txt

initramfs volumio.initrd gpu_mem=16 force_turbo=1 dtoverlay=iqaudio-dacplus

muzbox, 2016-02-22:
Fixed it, for anyone else who finds this issue

initramfs volumio.initrd gpu_mem=16 force_turbo=1 
dtoverlay=iqaudio-dacplus

the setting needs to be on its own line

closed 2016-03-25 by volumio