volumio/Volumio2 #219: Podcasts

kaffeeringe, 2016-02-08:
Hi! Thank you for your great work! It was really easy to set up Volumio and it sounds and looks great!

I love to listen to podcasts. How about support for podcasts in Volumio. 

I see there's a solution that adds podcasts to the mpd library: http://strobelstefan.org/?p=4513 But I don't want the podcasts among my music. 

How about you download the podcasts into a different folder and then stream them like they were radio stations?

fanciulli, 2016-02-16:
Hello @kaffeeringe 

the podcast support would be a great plugin for Volumio. Unfortunately it would need some effort on the UI side since i'm not sure its UI is possible with the tools we currently have.

@volumio what do you think about? I think it would be a great testing platform for a lugin SDK....

volumio, 2016-02-16:
Yep, let's go with the plugin SDK!

volumio, 2018-03-26:
Done with a plugin

closed 2018-03-26 by volumio

biva, 2018-03-27:
Good news @volumio ! Thank you
Which plugin is it?

DracoBlue, 2018-04-15:
https://github.com/ChrisPanda/volumio-podcast-plugin

kaffeeringe, 2018-04-25:
How do I install it? The plugin doesn't show up in my plugin list

biva, 2018-04-25:
Install an unofficial plugin
	1. SSH into machine
	2. Download zip using: wget <url>
	3. Uncompress in new folder
	4. Go into new folder
	5. Run: volumio plugin install

kaffeeringe, 2018-04-25:
Okay. Thank you. ;-)