volumio/Volumio2 #249: Album Art Quality

apinner, 2016-02-21:
I have been testing extensively over the past few days and one of the biggest annoyances so far is the awful quality of the album art. 

The images on desktop and the backgrounds on mobile are really really low resolution. It just ruins the whole look on mobile.

volumio, 2016-02-21:
We try to retrieve the albumart from disk if its present, if not we use lastfm API to retrieve 600x600 images (which should be fine...) So probably you have low res albumarts on your folders... try to delete those

apinner, 2016-02-21:
I don't have any album art in my music folders, i keep it very clean.

Its what is being downloaded.

Examples attached

[image]

[image]

volumio, 2016-02-21:
Can you check if they are embedded in id3 tag?

apinner, 2016-02-21:
So.. Bonobo actually had embedded artwork, which i have just removed from all tracks.

Loaded up in V2 and this is what it downloaded. 

[image]

Its a 174x174 image being blown up to 600x600. I know that it was downloaded because the embedded image i removed was 500x500.

volumio, 2016-02-21:
ok, so probably we want to look for higher res retrieval.
I'll look at it now ...

balbuze, 2016-03-16:
All downloaded image are stored in 174x174 pixels, that corresponds to the "large" image in last.fm.
In data/albumart/webmyartist/album/info.json "large" is writen. So it is coherent..
But It seems that /app/plugins/miscellaneous/album/albumart.js use "extralarge" with this it should be a 300x300 image displayed.
I did some unsuccessful test, even with "mega" value.
I read that http://www.albumartexchange.com/forums/viewtopic.php?f=1&t=1656 (see 3rd post)

apinner, 2016-03-17:
Nice bit of investigative work @balbuze. It definitely needs fixing, I'd rather have no album art than low quality blurry images.

balbuze, 2016-03-19:
I sent a PR to get album art in extra large quality. Mega is available but file bigger...

balbuze, 2016-03-25:
Just a thought : even with "extralarge" value, albumart are not very nice on big screen because it is a 300x300pix image (or less). Using "mega" download best file available but file can be very large (I saw one 3Mo) and unuseful because even on a full HD screen the image will 500x500px about. So maybe, the trick is to download "mega" image and resample it 500x500px before saving it. With that we get the best between appearance and weight.. What do you think of that ?

volumio, 2016-03-25:
Yes, this would be the best way for sure... Actually we are already doing some sort of resizing with the others... Let's try ...

apinner, 2016-05-04:
I am reopening this issue.

This is something i have noticed, but someone on the forum just pointed out as well. The new large album art slows down the queue page.

What we actually need is the nice clear large album art for the playback screen and mobile background and a small low res version that can be used in the queue where it is repeated multiple times.

martinchristophhess, 2016-05-05:
Hello everybody,
thank you for bringing this up!
What i'd really appreciate would be to be able to disable album art.
So a real standalone player wouldn't look like beeing incomplete and purists could reduce pixel "impact" to their brains ;-)
Have a nice day!
Martin

martinchristophhess, 2016-05-05:
Should the latter be an extra issue or wish eventually?

closed 2017-03-24 by volumio