volumio/Volumio2 #250: Hosted Interface

apinner, 2016-02-22:
I used Plex (https://plex.tv/) quite extensively for 99% of my media consumption. 

It is mainly controlled and configured through their webUI. I can either connect to the local instance of the plex media server using its local IP and port, or i can use https://app.plex.tv/web/app from anywhere and use https. I just have a random port open in my router, but it can also use upnp.

Obviously this is tied to a login. But is there any significant reason that volumio couldn't use the same setup with a hosted UI? I know volumio.local works on most machines, but android ignores this and you have to know the local IP. Connecting to say https://control.volumio.com/ would bypass all of this.

In theory then you could control several instances of volumio from pretty much anywhere.

jhoutman, 2016-02-22:
Its a nice idea.
You need a reverse dns lookup, and can put that into a router using UPnP. Altho, it's not the most desirable setup. A real hosted UI could work. The only thing is that creating a service like this, makes it a helluvalot different type of project.

We would need a paid service, cloud based on stuff like Amazon, a lot of IO, and ..well.. money :)

If you're creative enough, you can set it up locally. But not every router support this. And say if you're on DNS of Google and roll that out using DHCP, then you're already have a problem. 

Technically, it wouldn't be too hard to build a setup. The authentication mechanism is the most difficult part, and let alone all the security that has to come with that. Basically, you just connect to the hosted UI service, open a socket per device, tie it to a user login and use a hosted UI or even locally with a portal to choose device or something.

volumio, 2016-02-22:
It is something I'm thinking of since lot of time... And in the future I would really like to make it... I have a reverse proxy setup working and think can make it scalable to fit that scenario..
However lot of money is needed for the Volumio userbase size (60k+), so probably we would need to make it sort of premium feature...

apinner, 2016-02-22:
Well Plex have a Plex Pass feature that i assume pays the way for all that kind of thing. It adds additional features but everyone can use the hosted UI for free.

I wonder if you opened up a donations system how much you would get, I bet it would cover the costs of hosting a UI on Amazon or Azure.

volumio, 2016-02-22:
That may definetely be the way. After we complete volumio we'll think on it!

jhoutman, 2016-02-22:
Benefit is scaling and a good ddos protection indeed. But creating a proper user service.. pff man :) .. and which rules to comply to (european or USA law... patriot act, EU council privacy act... )

It is indeed a really interesting scenario.

apinner, 2016-02-22:
Wowser.. so much more complex than I thought!

Can of worms officially opened.

jhoutman, 2016-02-22:
Haha. Well, what we can do, is create a tutorial for people who want it, and perhaps a mini service using NodeJS page to select multiple ones. It's a tutorial for more advanced users. That would already help people getting this experience. With a disclaimer of course... ;-) (And the idea not to expose the interfaces to the outside world)

volumio, 2016-02-22:
Yep is definetely complex, but that's what makes it interesting...

apinner, 2016-02-23:
https://blog.filippo.io/how-plex-is-doing-https-for-all-its-users/

An interesting read.

volumio, 2016-02-23:
Clever.... Indeed very clever. 
Ok, we got caught into this... Let's think and make it happen ! (after the release...)

volumio, 2016-02-23:
I may have another solution (or a complementary one, if we want https) for this...

apinner, 2016-02-23:
I suppose https is a nice to have feature, but its not critical to a hosted UI working.

The main benefit is not having to mess around with IP addresses. A single entry point accessible from anywhere on any device for all of your instances of volumio is the main benefit.

jhoutman, 2016-02-23:
I would recommend definately using HTTPS. I've been initializing this for the company I work for with a shared web shop for 400 domains, which is a hell. The benefits for HTTPS is not just security, but the possibility of using web workers, HTTP/2 (not per se needed, but preferable) and of course compliancy.

The bad part is configuring load balancers, the computation load of encryption and managing expiration of certificates.

jshayden, 2016-12-15:
@volumio
Would you mind sharing your reverse proxy setup? I can connect through mine, but I get the spinning loading indication forever.

volumio, 2017-06-13:
Closing, because we will release something ad-hoc for this

closed 2017-06-13 by volumio