volumio/Volumio2 #253: Special characters in player name

Turbofoehn, 2016-02-25:
Hey everyone!
Great work so far!

I was wondering whether it was possible to have the player name include special characters such as _ä ö ü_ or even 📻 in the player name? And preserve capitalisation while we're at it?

As of now, they are simply omitted: 'Küche' (German for kitchen) becomes 'kche' etc.

While certainly not one of the most pressing issues, I believe would make Volumio even more user friendly.

FishTest, 2016-02-26:
Maybe it's mpd's problem,mpd will not output IDV3 tags with origin code.
if you use other mpd client ,such as mpc,It still has tag decode error

FishTest, 2016-02-26:
to solve this problem,
1,mpd output origin text without any convertion
2.volumio can add language selection,then covert origin text code to selected language(use libiconv).

volumio, 2016-02-26:
Sorry FishTest but your answers are unrelated... 

@Turbofoehn it's not possible to include special chars since the name will be written as hostname which doesn't accept special characters...

apinner, 2016-02-26:
Could this not be done with punycode? https://nodejs.org/api/punycode.html

closed 2016-06-27 by apinner