volumio/Volumio2 #254: Add everything to playlist

apinner, 2016-02-25:
Something I used to do often in 1.55 was add everything in my library to a playlist and just put the playlist on random.

I don't seem to be able to do that in 2.0. If I go Music Library > NAS > Music (which is the top level folder) and click the hamburger and Add to Playlist. It creates the playlist but its completely empty.

My Music is arranged:
Music
-Artist
--Album

I seem to have to go down to the album level for it to actually add to playlist.

FishTest, 2016-02-26:
one key add all music to mpd playing queue,expect that!

balbuze, 2016-02-26:
@apinner 
Yep you're right. The seems empty.
To play all tracks  I just click play on the top folder and it add all tracks in the queue. But it's long even with 423 artists in 255 albums ? and 2914 songs (sorry it's tracks ;-)) and my browser turn black several times before I can display the queue.

apinner, 2016-02-26:
Yeah, this is also the only place where the UI really seems to creep along and slows down.

I can add all to the queue by doing as you say, but not to a playlist.

balbuze, 2016-02-26:
In fact when you have a look in /data/playlist/ the playlist created by selecting a folder is empty. It's because it add nothing....
For the tracks I agree with you but in the my music it is still reported as "songs". To be changed !

closed 2016-04-03 by apinner

AlskiOnTheWeb, 2017-06-20:
Actually, I think the top level "Music", "Artists" etc should have a "Play All". More so, the individual artists should have a "Play All Similar" and use a depth of lastfm similar artists to fill out the list. For large libraries, the management UI for the queue is not very good.