volumio/Volumio2 #261: volume control

Nerz86, 2016-02-27:
Hey.
I have following issue.
When you change the volume (for example over ssh with "mpc volume 75") and then you go to the web page the volume will be only 25.
It seams that the web page did not load the current volume.

volumio, 2016-02-27:
No. Volumio changes (and retrieves) Volume directly on ALSA. MPD's volume is not used at all

closed 2016-02-27 by volumio

Nerz86, 2016-03-01:
All right, i understand. But how i have to change the volume over ssh. I tried it with "amixer cset numid=1 80%" but i have the same problem. When i go to the web page the volume will be changed to 25. Can you tell me how to change the volume directly on ALSA?
Thanks a lot.

balbuze, 2016-03-01:
And with alsamixer ? but your dac must have hardware mixer, or you have to create a virtual device "softplug" to be able to change to volume