volumio/Volumio2 #266: Raspberry Pi 3 support

davidpronk, 2016-03-02:
Now that the Raspberry Pi 3 is available it would be interesting to be able to test run Volumio 2 on it. I would be happy to help out by testing new builds on my Pi 3.

closed 2016-03-15 by davidpronk

storrgie, 2016-03-15:
Does this mean builds exist for the RPI3?

davidpronk, 2016-03-15:
Not to my knowledge. I closed the issue because I think it is not a specific Volumio 2 issue.

enavarro222, 2016-03-15:
Hi,

I'm using Volumio 1 in "prod", and plan to replace it by Volumio 2 on on rpi 3 (I'm not in a hurry).
So I tested Volumio 2 with a rpi 3, but the last image (Beta4 2016-03-03) do not boot with a rpi 3. Therefore I subscribe to this ticket to get some news, if a -rpi 3 compatible- image is released.

By the way, thanks for all the great works on Volumio !

davidpronk, 2016-03-15:
I managed to get the beta 4 running on a pi 3 by overwriting all *.dat and *.elf files with the ones from the official Raspbian 2016-02-26-raspbian-jessie-lite.img image.

volumio, 2016-03-15:
I completed the PI3 build yesterday... It works also with the integrated wireless chip.. I'm finishing tests and if everything is ok, then you'll have a new image soon..

davidpronk, 2016-03-15:
> I completed the PI3 build yesterday... It works also with the integrated wireless chip.. I'm finishing tests and if everything is ok, then you'll have a new image soon..

If you need help testing let me know...

donjuan201, 2016-03-17:
Hi guys, any idea when the new PI3 build will be made available publicly for testing?!

volumio, 2016-03-17:
I'm still testing it... But since you're so eager...
http://updates.volumio.org/pi/volumio/0.856/volumio-0.856-2016-03-17-pi.img.zip

davidpronk, 2016-03-18:
> I'm still testing it... But since you're so eager...
Great! The Pi3 is up and running on the 0.856 build.

SonosKiller, 2016-03-21:
Pretty neet ! Was running Volumio 1.55 on a Pi2 but didn't manage to get any wifi to work properly, as wild and crazy I am, let's try Volumio 2 (used the "2016-03-17" image), Wifi and Webradio seems to work nicely. My plan was to release my first SonosKiller "SK1" but I think I will postpone that and make a alpha release of my "SK2" instead. You are all blessed !

ebertland, 2016-03-21:
Does anybody have working built-in WiFi with the 0.856 build?

The build works for my RPi3 on Ethernet. If I switch to using the built-in WiFi, then I get extremely bad performance with ping latencies of several seconds. Using Raspbian, I get very good WiFi performance. If I'm the only one, I'll wait for the next beta and file a bug if I still see it.

volumio, 2016-03-21:
This may be due to power save mode... Can you report if this is enabled?

ebertland, 2016-03-22:
It was enabled. After disabling it, I get the same behavior. At first I thought it was packet loss, but if there is loss and retransmission, it is happening below layer 3. Let me know if you see anything wrong in the following log.

```
volumio@volumio:/etc/network$ /sbin/iwconfig
wlan0     IEEE 802.11bgn  ESSID:"ebertland"
          Mode:Managed  Frequency:2.462 GHz  Access Point: 30:B5:C2:B2:5F:A1
          Bit Rate=65 Mb/s   Tx-Power=1496 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=68/70  Signal level=-42 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

volumio@volumio:/etc/network$ ping -I eth0 10.1.1.1
ping: SO_BINDTODEVICE: Operation not permitted
volumio@volumio:/etc/network$ sudo ping -I eth0 10.1.1.1
PING 10.1.1.1 (10.1.1.1) from 10.1.1.21 eth0: 56(84) bytes of data.
64 bytes from 10.1.1.1: icmp_seq=1 ttl=64 time=0.332 ms
64 bytes from 10.1.1.1: icmp_seq=2 ttl=64 time=0.281 ms
64 bytes from 10.1.1.1: icmp_seq=3 ttl=64 time=0.279 ms
64 bytes from 10.1.1.1: icmp_seq=4 ttl=64 time=0.260 ms
64 bytes from 10.1.1.1: icmp_seq=5 ttl=64 time=0.269 ms
^C
--- 10.1.1.1 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 3996ms
rtt min/avg/max/mdev = 0.260/0.284/0.332/0.027 ms
volumio@volumio:/etc/network$ sudo ping -I wlan0 10.1.1.1
PING 10.1.1.1 (10.1.1.1) from 10.1.1.228 wlan0: 56(84) bytes of data.
64 bytes from 10.1.1.1: icmp_seq=1 ttl=64 time=1127 ms
64 bytes from 10.1.1.1: icmp_seq=2 ttl=64 time=1839 ms
64 bytes from 10.1.1.1: icmp_seq=3 ttl=64 time=2068 ms
64 bytes from 10.1.1.1: icmp_seq=4 ttl=64 time=2470 ms
64 bytes from 10.1.1.1: icmp_seq=5 ttl=64 time=2459 ms
64 bytes from 10.1.1.1: icmp_seq=6 ttl=64 time=1458 ms
64 bytes from 10.1.1.1: icmp_seq=7 ttl=64 time=1707 ms
64 bytes from 10.1.1.1: icmp_seq=8 ttl=64 time=706 ms
64 bytes from 10.1.1.1: icmp_seq=9 ttl=64 time=32.3 ms
64 bytes from 10.1.1.1: icmp_seq=10 ttl=64 time=260 ms
64 bytes from 10.1.1.1: icmp_seq=11 ttl=64 time=53.8 ms
64 bytes from 10.1.1.1: icmp_seq=12 ttl=64 time=22.1 ms
64 bytes from 10.1.1.1: icmp_seq=13 ttl=64 time=570 ms
64 bytes from 10.1.1.1: icmp_seq=14 ttl=64 time=101 ms
64 bytes from 10.1.1.1: icmp_seq=15 ttl=64 time=16.0 ms
64 bytes from 10.1.1.1: icmp_seq=16 ttl=64 time=197 ms
64 bytes from 10.1.1.1: icmp_seq=17 ttl=64 time=1242 ms

```

volumio, 2016-03-22:
I get much better results...

.
64 bytes from 192.168.1.55: icmp_seq=1 ttl=64 time=7.56 ms
64 bytes from 192.168.1.55: icmp_seq=2 ttl=64 time=7.01 ms
64 bytes from 192.168.1.55: icmp_seq=3 ttl=64 time=8.40 ms
64 bytes from 192.168.1.55: icmp_seq=4 ttl=64 time=106 ms
64 bytes from 192.168.1.55: icmp_seq=5 ttl=64 time=9.27 ms
64 bytes from 192.168.1.55: icmp_seq=6 ttl=64 time=8.88 ms
64 bytes from 192.168.1.55: icmp_seq=7 ttl=64 time=8.30 ms
64 bytes from 192.168.1.55: icmp_seq=8 ttl=64 time=9.02 ms
64 bytes from 192.168.1.55: icmp_seq=9 ttl=64 time=7.98 ms
64 bytes from 192.168.1.55: icmp_seq=10 ttl=64 time=8.86 ms
64 bytes from 192.168.1.55: icmp_seq=11 ttl=64 time=60.3 ms
64 bytes from 192.168.1.55: icmp_seq=12 ttl=64 time=10.9 ms
64 bytes from 192.168.1.55: icmp_seq=13 ttl=64 time=8.64 ms
64 bytes from 192.168.1.55: icmp_seq=14 ttl=64 time=11.1 ms
64 bytes from 192.168.1.55: icmp_seq=15 ttl=64 time=9.33 ms
64 bytes from 192.168.1.55: icmp_seq=16 ttl=64 time=8.60 ms
64 bytes from 192.168.1.55: icmp_seq=17 ttl=64 time=8.82 ms
64 bytes from 192.168.1.55: icmp_seq=18 ttl=64 time=6.37 ms

I am investigating...

sbkivitz, 2016-03-25:
**Raspberry Pi 3 Wireless issues**
I installed  the 0.856 build on my raspberry pi 3 and had some initial difficulties getting wlan0 to receive an IP address via DHCP.  The router saw that the MAC address of the pi's wlan was associated with it, but didn't give it an IP.

I resolved/worked around this by assigning the pi's wlan a static IP by MAC address on my DD-WRT router.  Google for those instructions - it's under the services tab in the router GUI.

DJKNK, 2016-04-07:
Any news for Volumio on R-Pi 3 ?

davidpronk, 2016-04-07:
The latest release candidate runs on the Pi3:
http://updates.volumio.org/pi/volumio/0.861/volumio-0.861-2016-03-25-pi.img.zip

DJKNK, 2016-04-07:
Ok thanks ;)

DJKNK, 2016-04-07:
Well, doesn't work ... sorry, I'm completely new to this kind of things ...
[image]

torbjornvatn, 2016-04-16:
I also have problems with `Exceeded CRDA call max attempts` when trying to boot the RC image

rlucian, 2016-04-20:
I am following up on the improvements and it looks promissing...
So, any chance a RC2 would be released?

ZeroQool007, 2016-04-29:
I've got the same problem 

> Exceeded CRDA call max attempts

I use this built: http://updates.volumio.org/pi/volumio/0.861/volumio-0.861-2016-03-25-pi.img.zip

volumio, 2016-04-29:
Should be fixed with this:
https://github.com/volumio/Build/commit/70e12e6eeccc16e00c36167b8196af3c639ba5bd

ZeroQool007, 2016-04-29:
thx, where can I download the full image? thx

volumio, 2016-04-29:
Yet to be built. If you want to fix yours do:

apt-get update
apt-get -y install wireless-regdb crda

Let me know how it goes

Thomas0774, 2016-05-07:
it works!
stable connection between pi3 and router.

DJKNK, 2016-05-18:
Unable to fetch the files with the get update command ...

rbeekink, 2016-05-21:
@Thomas0774 how did you manage to do this. I'm a bit lost on the volumio/Build@70e12e6...no idea where to make this change. Hope you can help...

rbeekink, 2016-05-21:
Ok! Finally got it working on Raspberry 3 with Volumio and Hifiberry. On Ethernet. Will try WiFi tomorrow. Sound is amazing!

JeroenfNL, 2016-05-24:
Hello members, I have the same quenstion as 'rbeekink' asked Thomas.

@rbeekink, can you tell me how you did the job? 
@Thomas0774: is there allready a working image?

All suggestions are welcome, 
thanks in advance,  Jeroen

volumio, 2016-05-24:
Sorry guys I don't get what doesn't work for you... It means that you can't connect to wifi?

rbeekink, 2016-05-24:
I initially also got the CRDA call max attempts error and followed the comments of member volumio. This worked, however, it turned out the sound was coming via HDMI and not via HifiBerry. I then looked on the HifiBerry site and via the HifiBerry Installer you can compile an image for Volumio [ https://www.hifiberry.com/guides/hifiberry-installer/ [https://www.hifiberry.com/guides/hifiberry-installer/] ] This still didn’t work. I then again entered comments of member volumio:
In terminal: apt-get update apt-get -y install wireless-regdb crda
But I also needed to edit config.txt : dtoverlay=pi3-disable-bt
Now it works, but still haven’t got a wifi connection. It says: no adapter present? Maybe need to update..
via CloudMagic for Mac [https://cloudmagic.com/k/d/mailapp?ct=dx&cv=7.12.31&pv=10.10.5&source=email_footer_2]

boskodin, 2016-06-01:
I know this is not the topic of the issue, but I'm wondering will there be bluetooth reciever option in the final build? I'm desperate for that option

apinner, 2016-06-01:
I doubt very much it will be in the next release. Not sure about the final release, I am sure it's on the roadmap somewhere, probably as a plugin.

volumio, 2016-06-01:
It won't be on next release, but as apinner suggested it's on the roadmap, probably as a plugin...

Tiefschneegiftler, 2016-10-06:
Sorry for bumping this, but is there a working solution for the HiFiBerry on the Pi 3 using its integrated wifi running Volumio?

With typing:
apt-get update
apt-get -y install wireless-regdb crda
I only get errors...

marioslink1, 2017-05-03:
hello. when I go to put plugin, the page is reload..... its not possible to add plugin. why??