volumio/Volumio2 #267: DSD

hicaoc, 2016-03-02:
DOP
Native DSD

balbuze, 2016-03-03:
I've got DSD files and I can play them in Volumio2

hicaoc, 2016-03-04:
@balbuze 
    Yes , the decoding to PCM can be played , but my DAC supports native DSD and DoP, currently no DoP and native DSD configuration on the UI 。I modified mpd.conf file to add options to achieve DoP play DSD 。

budisaka, 2016-07-10:
@hicaoc 
how to edit mpd.conf..?
I've got error message : Error: no "edit" mailcap rules found for type "application/octet-stream"

warrenjmcdonald, 2016-08-06:
sudo apt-get install vim
vim /etc/mpd.conf

In order to be able to stream dsd over pcm you should add the following to mpd.conf your device section:

dsd_usb         "yes"

example:
 
audio_output {
        type            "alsa"
        name            "Rockna - Audiobyte Silver Dragon"
        device          "hw:0,0"
        dsd_usb         "yes"
} 

Save 
restart volumio

volumio, 2017-06-02:
Implemented

closed 2017-06-02 by volumio