volumio/Volumio2 #272: Latest Beta Questions

jncraig, 2016-03-07:
Should I expect the latest beta to be ready for external sound cards, including USB soundcards?

How about WiFi?  Can I configure that from a terminal window?

jncraig, 2016-03-07:
I can connect to Volumio using WiFi from the Network page.

I see flashing activity on the USB device, and ifconfig confirms I'm connected.  The Network page also shows a connection.

But, I cannot connect to the the WiFi IP address UNLESS the ethernet cable is also connected.

Looking at traffic with Fiddler shows some http requests aimed at the ethernet IP address even when I am connected to the WiFi IP address and refresh the page.

apinner, 2016-04-03:
WiFi and DAC was added in RC1, closing.

closed 2016-04-03 by apinner