volumio/Volumio2 #273: Version 493 does not update UI for My Music while database is updating

piercer, 2016-03-07:
Hi,

the real-time update of the 'My Music' no longer works while the database update is running.

Doing a 

sudo systemctl status mpd

shows that MPD is updating its database, but this is not reflected in the UI

piercer, 2016-03-07:
I've now just seen that once the number of songs/tracks and artists exceeds the number there was before it starts to update.

I am sure it used to reset to zero when rescanning, but did not see that this time.

If this is the desired behaviour then feel free to close this ticket.

apinner, 2016-04-03:
I don't recall it ever resetting to zero, i think this is the desired behaviour.

Will close but feel free to reopen if necessary

closed 2016-04-03 by apinner